genodelabs/genode #4271: depot: copying of package content fails without hint

skalk, 2021-09-29:
Whenever a package with empty content is part of a run-script dependency (e.g., because the package does not compile anything for the given architecture), the run-script fails without any hint, which package is affected at all. The error look like:

```
no files matched glob pattern "*"
    while executing
"glob -directory $dir *"
    (procedure "_copy_directory_content_to_run_dir" line 8)
    invoked from within
"_copy_directory_content_to_run_dir "[depot_dir]/$subdir""
    (procedure "import_from_depot" line 12)
    invoked from within
"import_from_depot [depot_user]/src/[base_src] \
                  [depot_user]/pkg/sculpt-[board]"
    (file "/home/sk/src/genode/repos/gems/run/sculpt.run" line 12)
    invoked from within
"source $include_name"
    ("foreach" body line 6)
    invoked from within
"foreach include_name [get_cmd_arg --include ""] {
  # first check if the include name is absolute
  if {[string first "/" $include_name] == 0} {
    puts ..."
    (file "/home/sk/src/genode/tool/run/run" line 1062)
make: *** [Makefile:396: run/sculpt] Error 1
```

I would like to suppress this error, because in some cases it might be possible that an empty package is okay for this architecture. On the other hand, if content is missing I see it at a later point, where the missing content is recognizable. Other opinions?

skalk, 2021-09-29:
Commit 6d36c6c suppresses the failure.

nfeske, 2021-09-29:
Thank you. Merged to staging.

commit 909d2b03 (2021-09-29): depot: suppress error when copying empty package

Instead of failing, ignore empty packages when copying over
depot packages to a run-script environment.

Fix #4271