genodelabs/genode #4272: dde_linux: possible deadlock in `Device::irq_mask`

nfeske, 2021-09-29:
The `Device::irq_mask` implementation of the dde_kit destructs the IRQ handler, which is a signal handler:

```
  _for_each_irq([&] (Irq & irq) {
    if (irq.number != number)
      return;
    irq.handler.destruct();
  });
```

Unfortunately, there is the chance that this code is executed in the context of the IRQ signal handler itself, ultimately ending up in a deadlock at https://github.com/genodelabs/genode/blob/master/repos/base/src/lib/base/signal_common.cc#L218.

We should reconsider the destruction of the `Io_signal_handler`. @skalk do you see an alternative implementation of `irq_mask` that would preserve the `Io_signal_handler`?

tomga, 2021-09-29:
For some time I'm hunting a problem (for which I asked on ML thinking that interrupts are not processed) that acquiring lock on `_contexts_mutex` blocks infinitely in usb driver on Raspberry Pi 2. May it be possible that it is somehow the same issue?

This lock mutex attempt is in my case in: https://github.com/genodelabs/genode/blob/64e2912a2ff1e8c2ad171ea1efb96ce94dd01aa9/repos/base-hw/src/lib/base/signal_receiver.cc#L100

I was going to attempt to diagnose it further tonight but as you understand it much better and are already hunting issues here you'll be able to easily solve my issue too. I tried to find a place where mutex is acquired but until now I failed to find it even with adding logs in `Mutex::Guard`.

Let me know in case it is not the same issue. I won't pollute this issue more with unrelated problem.

skalk, 2021-09-29:
Mmh that is strange. Typically Linux only masks the interrupt when it gets released. It is strange that this is done inside the top-halve of the interrupt handling. But I assume you don't opened the issue for hypothetical reasons, but instead observed this behaviour.
Not closing the interrupt slightly changes the hardware semantic in contrast to the original one. Of course we can keep the state of the interrupt being (un-)masked, and whether it triggered and got aknowledged. In that case we would flag that the interrupt occurred, and when it got unmasked again, we have to deliver it. But I see upcoming problems with drivers that cannot handle wrongly asserted interrupts well. Because it could happen that an interrupt occurs while it should be masked, then the device actually goes into a state where the interrupt line is quite again, but we deliver it nevertheless.

Alternatively, we could detect during the "irq_mask" that we in an interrupt signal handler, do not execute the irq_mask but put it into a tasklet, workqueue, or whatever, and then after the interrupt signal handler it gets scheduled immediatedly. To me this seems more robust, but maybe I oversee something.

skalk, 2021-09-29:
Okay after @tomga comment, I've looked inside the legacy DDE Linux interrupt handling, and I have to re-think my former assessment. We can do it similar to the former approach by flagging a pending interrupt when it occurred in masked state. When it gets unmasked again the interrupt gets acknowledged without asserting an interrupt to the Linux code. If the interrupt reason is still valid, it will trigger again immediately, so nothing gets lost.
I can provide a fix for that.

@tomga I do not think it is related to your issue. The behaviour @nfeske describes is related to the "new" DDE Linux. In the former emulation environment there was no dynamic destruction of the interrupt signal handler happening. I'm sorry.

skalk, 2021-09-29:
@nfeske I've found a more easy solution for the issue. For no good reasons the Io_signal_handler was part of the Constructible, although it is enough to enclosure the Platform::Irq session with a Constructible to get the same semantic. Please try out commit 6bb590e

nfeske, 2021-09-29:
Thank you @skalk for attending this issue so quickly. The issue triggered quite sporadically, once every few days. I cherry-picked your commit and will keep testing it.

commit a1719087 (2021-09-30): dde_linux: preserve Io_signal_handler for IRQs

Do not construct/destruct signal handlers for interrupts dynamically,
but only the Irq session to prevent a deadlock.

Fix #4272