genodelabs/genode #4284: First steps to make usage of the Gpu session more robust

cnuke, 2021-10-05:


cnuke, 2021-10-05:
The following commits are required to prevent resource shortage - either the `intel_gpu_drv` or the `platform_drv` will run out of caps - when using the _Gpu_ session on Sculpt:

* 199cb8b203 - platform_drv: check quota before allocating buffer

  Adds a check to `alloc_dma_buffer` to prevent the the platform driver from issuing a resource request because its internal
  PD session is running out.

* ed1fce9de2 - gpu: reflect CAP shortage during buffer management

  Allow for reflecting a _Out_of_caps_ situation back to the client so that it might be handled gracefully.

* 80cea2f8b9 - gpu/intel: deal with insufficient amount of CAPS

  This commit adds handling of resource shortage to the multiplexer by upgrading the _Platform_ session on user allocation explicitly.

  (To be honest I am not really happy with the commit as it is really just a band-aid and deals with issue of separating allocations of client data and meta-data allocations in the multiplexer on behalf of the client rather crudely. We need to schedule a proper overhaul of the component.)

* 11d1070b23 - squash 'gpu/intel: deal with insufficient amount of CAPS' (libdrm/iris)

  As the commit subject suggests, could be part of the first one and is there for easier review.

ssumpf, 2021-10-11:
@nfeske: 04cb157 uses `Region_map_clients` instead of `Io_mem_connections` for Aperture mapping.

commit d9e2b0f7 (2021-10-06): gpu/intel: deal with insufficient amount of CAPS

'Out_of_ram' was so far the only exception a client had to deal with
during buffer managment. Allocating memory, however, does not only
consume RAM quota but CAP quota as well.

This commit tries to mitigate that shortcoming by reflecting the
'Out_of_caps' state back to the client. Furthermore it allows for
resource accounting on certain client allocations, e.g. buffers.

Fixes #4284.