genodelabs/genode #4311: tool/run: streamline board-specific definition of qemu arguments

jschlatow, 2021-11-02:
In context of #4280, which moves the Zynq-board support (including zynq_qemu) to an extra repository, I had the idea to put the definition of qemu arguments into a separate file next to `image_link_address` and `arch`. After an offline discussion with @skalk, we decided to address and discuss this in separate issue since it would be nice to apply this scheme to all `*_qemu` boards. 

The benefit of this is that we could move the definition of qemu_args to board-support repositories and potentially eliminate a couple of case distinctions from the run tool.
On the other hand, there are a few boards (pbxa9, virt_qemu) that currently require special treatment (see `tool/run/power_on/qemu`). For example, virt_qemu exists for arm_v8a and arm_v7a with the consequence that simply adding the qemu command line arguments to a single `qemu_args` file would not suffice.

@chelmuth, @nfeske: What's your opinion on this? Do you think it is worth going down this road?

jschlatow, 2021-11-02:
Commit c5ce41a serves as a blueprint. Note, that it only contains the necessary changes for zynq_qemu and doesn't apply the scheme to other boards yet.

nfeske, 2021-11-02:
> What's your opinion on this? Do you think it is worth going down this road?

That would be a very welcome improvement. I think that using the `board` directory for the qemu arguments is an obvious and elegant approach but I haven't seen this connection so far. Cool!

chelmuth, 2021-11-02:
I'd love this implemented.

jschlatow, 2021-11-03:
@nfeske, @chelmuth: I just pushed the corresponding commit series.

Commit 1c1bd76 improves how the run tool parses the provided -m arguments. It also suggests increasing the amount of memory for pc to 800M as this is a) required by okl4 and b) by depot_autopilot.run.

Commit 6094c3a applies the main changes to move the qemu arguments into a file. I came up with a simple syntax to restrict the additon of arguments to certain specs (required for virt_qemu). What I find particularly elegeant about this change is that the presence of the qemu_args file can be used for properly implementing `is_qemu_available`.

Commit 2f40bcc goes a step further and also moves the netdev instantiation into the qemu_args file. The side-effect is that there is no need to call `append_qemu_nic_args` in run scripts anymore unless forwarding rules are needed.

jschlatow, 2021-11-05:
@chelmuth, as discussed offline, I pushed the new commits 01f0320 and 9583da8 that shall replace the corresponding commits on staging.

nfeske, 2021-11-30:
Fixed in master.

closed 2021-11-30 by nfeske