genodelabs/genode #4312: riscv_qemu: move to genode-riscv repository

chelmuth, 2021-11-04:
In line with #4301, #4280, and genode-allwinner, Qemu-specific riscv files should be moved to https://github.com/ssumpf/genode-riscv/. The depot archive for drivers_nic-riscv_qemu can already be found there.

ssumpf, 2021-11-04:
@chelmuth: Shall I do it?

chelmuth, 2021-11-04:
@ssumpf Yes, that would be excellent. I expect we'll return to the riscv topic soon, which will be the right time to address this issue.

ssumpf, 2021-12-10:
bedb2e8, 5cb9bb4, ce85b5f, dba2a81, f27b44f are the commits for staging. f27b44f is optional because it is MiG-V specific.

chelmuth, 2021-12-14:
The entire series is now merged to staging and the nightly CI was armed. Take cover!

ssumpf, 2021-12-15:
@chelmuth: 4f8eb2b and 6580646 address `build-packages` errors from last night for riscv.

nfeske, 2021-12-16:
A few network-related tests of the depot autopilot currently fail. E.g., nic_loopback:

```
[init -> test-nic_loopback] --- NIC loop-back test ---[0m[0m
[init -> test-nic_loopback] -- starting roundtrip test --[0m[0m
[init -> test-nic_loopback] start iteration 0 with pattern 'a'[0m[0m
[init -> test-nic_loopback] allocated tx packet offset=32832, size=100[0m[0m
[init -> test-nic_loopback] received rx packet offset=32834, size=100[0m[0m
[init -> test-nic_loopback] Error: _consume_and_compare_packet: unexpected offset of received packet[0m[0m
```

I guess that it related to commit feec919d7afef2ea760108e55d852399576ddbd6. @ssumpf could you have a look?

ssumpf, 2021-12-17:
> A few network-related tests of the depot autopilot currently fail. E.g., nic_loopback:
> 
> ```
> [init -> test-nic_loopback] --- NIC loop-back test ---�[0m�[0m
> [init -> test-nic_loopback] -- starting roundtrip test --�[0m�[0m
> [init -> test-nic_loopback] start iteration 0 with pattern 'a'�[0m�[0m
> [init -> test-nic_loopback] allocated tx packet offset=32832, size=100�[0m�[0m
> [init -> test-nic_loopback] received rx packet offset=32834, size=100�[0m�[0m
> [init -> test-nic_loopback] Error: _consume_and_compare_packet: unexpected offset of received packet�[0m�[0m
> ```
> 
> I guess that it related to commit [feec919](https://github.com/genodelabs/genode/commit/feec919d7afef2ea760108e55d852399576ddbd6). @ssumpf could you have a look?

@nfeske: 9a91574 addresses this issue.

nfeske, 2021-12-17:
Fixed in master.

closed 2021-12-17 by nfeske

nfeske, 2021-12-17:
Thanks @ssumpf for the fix. I added your commit to staging.