genodelabs/genode #4314: nova: support transfer of IA32_TSC_AUX MSR for vCPUs

chelmuth, 2021-11-04:
The 32-bit MSR is returned by `rdtscp` in `ecx` register and used to detect the CPU ID the timestamp was taken on.

chelmuth, 2021-11-04:
@alex-ab the current implementation suggest that `tsc_aux` in UTCB has 64 significant bits while the spec states a 32-bit MSR. Should we change this?

alex-ab, 2021-11-05:
From my reading the MSR is 64bit, but solely "the lower" 32bit are usable by rdtscp:

```... the EAX register is loaded with the low-order 32 bits of the IA32_TSC MSR ... ```

The documentation is a bit fussy on what happens to the higher bits, mainly some hints suggests that they are not used/reserved.

Intel doc:  Intel® 64 and IA-32 Architectures
Software Developer’s Manual, May 202, Table 2-2. IA-32 Architectural MSRs (Contd.):

```
C000_0103H IA32_TSC_AUX, 31:0 AUX: Auxiliary signature of TSC, 63:32 Reserved
```

AMD doc specifies a 64bit value as reset value:

```
C000_0103h TSC_AUX The RDTSCP instruction copies the value of this
MSR into the ECX register. Reset value: 0000_0000_0000_0000h
```

chelmuth, 2021-11-05:
Thanks for the investigation. It seems MSRs are always 64-bit but may have less significant bits (in case of TSC_AUX only 32 bits). Values of insignificant bits are undefined on `rdmsr` but `rdtscp` ensures bits [63:32] of `rcx` are cleared.

commit 0c3f7be2 (2021-11-04): vbox5/6: set ia32_tsc_aux for rdtscp usage

Fixes #4314
commit d9018db1 (2021-11-04): vbox5/6: set ia32_tsc_aux for rdtscp usage

Fixes #4314
closed 2021-11-30 by chelmuth