genodelabs/genode #4317: Simplify `List_model` interface

nfeske, 2021-11-05:
The so-called `List_model` (https://github.com/genodelabs/genode/blob/master/repos/base/include/util/list_model.h) has become an established pattern in Genode components that maintain an internal data model for XML input data.

By extensively using the mechanism, I found myself repeatedly implementing a wrapper with simpler interface. Hence, I'd like to promote this interface to eventually become the actual list-model interface. The new interface consists of a single function template that relies on a few functions provided by the managed node type. It thereby removes the need to implement a custom `Policy` class. The idea was first introduced in the sandbox library: https://github.com/genodelabs/genode/blob/master/repos/os/src/lib/sandbox/update_list_model.h

To allow a smooth migration to the simpler interface, I'd like to first add the `update_list_model_from_xml` function to the existing util/list_model.h header. Then we can update the components step by step, followed by removing the original API.

nfeske, 2021-11-05:
Commit https://github.com/genodelabs/genode/commit/ff52b846aacf364b6e0b47278dc16457ea6df222 extends util/list_model.h with the new `update_list_model_from_xml` function.

Commit https://github.com/genodelabs/genode/commit/36743c9a09ea498570d9b66bc440295b0bef4460 adjusts the sandbox library to use the variant provided by util/list_model.h now.

Note the addition of the `NODE::type_matches` class function as a policy for the transformation. It replaces the former `POLICY::node_is_element` hook. This change is needed to eventually remove the notion of a policy class from the public interface, yet still allow the expression of the policy.

nfeske, 2023-10-26:
@skalk commit https://github.com/genodelabs/genode/commit/a2e83ca068a7080caef0bad1ba56e4c664259e09 updates the platform driver.

nfeske, 2023-10-26:
@skalk and commit https://github.com/genodelabs/genode/commit/81283050920c2e773a93faa703a8936f92f0e7b6 adjusts pci_decode. I hope you are fine with the changes.

nfeske, 2023-10-26:
Commit https://github.com/genodelabs/genode/commit/80199ffe6a2fe85559b4da9a0393e860dfb47066 crosses the ARM VMM from the list.

nfeske, 2023-10-26:
@m-stein, commit https://github.com/genodelabs/genode/commit/ddc86a612ee16858232ef959493f2329cfec517f adjusts WireGuard as the last remaining user of the original policy-class pattern. I had to add a public accessor `Config_model::Peer_update_policy::public_key_b64` to perform the sanity check during destroy and wondered if the `Peer` could be turned in a plain struct with all members public?

m-stein, 2023-10-26:
@nfeske I have no objections.

nfeske, 2023-10-27:
Thanks @m-stein for the immediate response. Commit https://github.com/genodelabs/genode/commit/55bf5782d7fbfca3c298fc57c462afa2857ed328 tightens the code. Do you approve?

m-stein, 2023-10-27:
Looks fine.

nfeske, 2023-10-27:
The commit series above wraps up the transition away from the old policy-based interface. Commit https://github.com/genodelabs/genode/commit/759b700811866e26429d0a7631cf655702d40c5a contains the actual list-model simplification.

To finalize the work on this issue, I plan to remove the `update_list_model_from_xml` function and leave only the `List_model::update_from_xml` method.

nfeske, 2023-10-28:
Commit https://github.com/genodelabs/genode/commit/96c9cb8f380d626fd78b1fcf972d426bc932c0a7 consistently replaces the `update_list_model_from_xml` function by the `List_model::update_from_xml` method.

While being at it (changing the API), commit https://github.com/genodelabs/genode/commit/600d1db88860d3c1eb636fdf6788b02dd12bd5aa renames `apply_first` to `with_first`, which is more consistent with other parts of Genode's API (e.g., `Dictionary`).

skalk, 2023-10-30:
@nfeske: I've no objections regarding the VMM, platform_drv, and pci_decode changes. Thank you for the simplification.

commit 759b7008 (2023-10-27): Simplify 'List_model'

This patch replaces the original policy-based 'update_from_xml' by a new
method that takes three functors for creating, destroying, and updating
elements as arguments. XML nodes are associated with their corresponding
internal data models by annotating the element type with the
'type_matches' class function and the 'matches' method.

The patch also improves safety by enforcing that list-model can never be
copied.

Fixes #4317
commit ecf6bd71 (2023-10-28): Simplify 'List_model'

This patch replaces the original policy-based 'update_from_xml' by a new
method that takes three functors for creating, destroying, and updating
elements as arguments. XML nodes are associated with their corresponding
internal data models by annotating the element type with the
'type_matches' class function and the 'matches' method.

The patch also improves safety by enforcing that list-model elements can
never be copied.

Fixes #4317
commit 56e9b0ea (2023-11-02): Simplify 'List_model'

This patch replaces the original policy-based 'update_from_xml' by a new
method that takes three functors for creating, destroying, and updating
elements as arguments. XML nodes are associated with their corresponding
internal data models by annotating the element type with the
'type_matches' class function and the 'matches' method.

The patch also improves safety by enforcing that list-model elements can
never be copied.

Fixes #4317