genodelabs/genode #4318: DNS settings for libc application

sergey-platonov, 2021-11-08:
Hello all

We have noticed that for the `libc` application DNS settings will be only applied if we create the `socket` directory in the root of the `vfs` node:
```
<libc socket="/socket" ... />
<vfs>
    <dir name="socket"> <lwip dhcp="yes"> </dir>
</vfs>
```  

We used to place `socket` dir in the `/dev` and in that case, DNS does not work.

Here is a runscript that demonstrates the problem (it's merely a copy of `genode-world/run/fetchurl.inc` with few simplifications):
<details>
<summary>runscript</summary>

```
create_boot_directory

set depot_archives { }

lappend depot_archives [depot_user]/src/[base_src]
lappend depot_archives [depot_user]/pkg/[drivers_nic_pkg]
lappend depot_archives [depot_user]/src/curl
lappend depot_archives [depot_user]/src/fetchurl
lappend depot_archives [depot_user]/src/init
lappend depot_archives [depot_user]/src/libc
lappend depot_archives [depot_user]/src/libssh
lappend depot_archives [depot_user]/src/nic_router
lappend depot_archives [depot_user]/src/openssl
lappend depot_archives [depot_user]/src/vfs
lappend depot_archives [depot_user]/src/vfs_lwip
lappend depot_archives [depot_user]/src/zlib

install_config {
<config>
	<parent-provides>
		<service name="CPU"/>
		<service name="IO_MEM"/>
		<service name="IO_PORT"/>
		<service name="IRQ"/>
		<service name="LOG"/>
		<service name="PD"/>
		<service name="RAM"/>
		<service name="RM"/>
		<service name="ROM"/>
	</parent-provides>

	<default-route>
		<any-service> <parent/> <any-child/> </any-service>
	</default-route>

	<start name="linux_nic_drv" ld="no" caps="100">
		<resource name="RAM" quantum="4M"/>
		<config tap="tap0" mac="02:00:00:00:00:00"/>
	</start>

	<start name="timer" caps="100">
		<resource name="RAM" quantum="1M"/>
		<provides> <service name="Timer"/> </provides>
	</start>

	<start name="nic_router" caps="200">
		<resource name="RAM" quantum="10M"/>
		<provides>
			<service name="Nic"/>
			<service name="Uplink"/>
		</provides>
		<config verbose_domain_state="yes" verbose_packets="no">
			<domain name="uplink" interface="10.0.2.2/24" gateway="10.0.2.1">
				<nat domain="downlink" tcp-ports="16384" udp-ports="16384"/>
			</domain>

			<domain name="downlink" interface="10.0.3.1/24">
				<dhcp-server ip_first="10.0.3.2" ip_last="10.0.3.2">
					<dns-server ip="8.8.8.8"/>
					<dns-server ip="1.1.1.1"/>
				</dhcp-server>
				<tcp dst="0.0.0.0/0"><permit-any domain="uplink" /></tcp>
				<udp dst="0.0.0.0/0"><permit-any domain="uplink" /></udp>
			</domain>
			<policy label_prefix="fetchurl" domain="downlink"/>
			<policy label_prefix="linux_nic_drv"  domain="uplink"/>
		</config>
	</start>

	<start name="fetchurl" caps="500">
		<resource name="RAM" quantum="32M"/>
		<config>
			<libc rtc="/dev/rtc" stdin="/dev/null" stdout="/dev/log" stderr="/dev/log"
				  socket="/dev/socket" rng="/dev/random"/>
			<vfs>
				<dir name="dev">
					<log/> <null/> <inline name="rtc">2021-11-01 00:00</inline>
					<inline name="random">01234567890123456789</inline>
				        <dir name="socket"> <lwip dhcp="yes"/> </dir>
				</dir>
			</vfs>
			<fetch url="http://genode.org/about/LICENSE" path="/dev/log" />
		</config>
	</start>
</config>

}

import_from_depot $depot_archives

build_boot_image {}

append qemu_args " -nographic "
run_genode_until {child "fetchurl" exited} 60
```

</details>

Running it we have an error:
```
[init -> fetchurl] Error: Couldn't resolve host name (code=0), failed to fetch http://genode.org/about/LICENSE
```

If we move `/dev/socket` to `/socket` it works.

The runscript uses `lwip`, but the same stands for the `lxip`.

sergey-platonov, 2021-11-08:
Sure, glad to help

chelmuth, 2021-11-24:
@sergey-platonov I hope I did not any commit you provided. Up to now no fix was applied to staging.

sergey-platonov, 2021-11-24:
No, sorry for the delay. I was participating in the conference and had less spare time than I anticipated. Thank you for the fix!

chelmuth, 2021-11-25:
> > Up to now **no fix** was applied to staging.
> 
> No, sorry for the delay. I was participating in the conference and had less spare time than I anticipated. Thank you for the fix!

Don't get me wrong, I'm still looking forward to **your** fix ;-)

sergey-platonov, 2021-12-14:
Hi @chelmuth 
I have pushed a couple of commits and created a pull request #4343. Looking forward to your feedback :)

commit 72bba6bc (2021-12-20): libc: nameserver file in configured socket directory

The path for the `nameserver` file was fixed to `/socket/nameserver`.
So, if the socket directory was configured differing from `/socket`, DNS
did not work. Now the default path for the `nameserver` takes into
account the path configured in the `socket` libc config attribute.

Fixes #4318
Fixes #4343
commit 81acd465 (2022-01-19): libc: nameserver file in configured socket directory

The path for the `nameserver` file was fixed to `/socket/nameserver`.
So, if the socket directory was configured differing from `/socket`, DNS
did not work. Now the default path for the `nameserver` takes into
account the path configured in the `socket` libc config attribute.

Fixes #4318
Fixes #4343
closed 2022-01-19 by chelmuth