genodelabs/genode #4319: Avoid use of `which` utility

nfeske, 2021-11-09:
Debian recently declared the `which` utility as deprecated. As Genode's tools use this command at several places - in particular to probe for the installation of certain dependencies - we have to adjust our tools.

The following article covers the background information behind this change: https://hynek.me/til/which-not-posix/

nfeske, 2021-11-09:
I tried to fix this issue via commit https://github.com/genodelabs/genode/commit/f3a78774474edee7d6610227e04589dad4a80c2e

To my surprise, the following does not work in the build.mk Makefile when executed on my Xubuntu installation:

```
export SHELL := $(shell command -v bash)
```

The resulting value of `SHELL` is empty. This can be reproduced withe the following `Makefile`:

```
export SHELL := $(shell command -v bash)
$(warning SHELL=$(SHELL))
default:
```

However, by explicitly spawning the sub shell, it works:
```
export SHELL := $(shell sh -c "command -v bash")
```

In the remainder of the Makefile (once `SHELL` is set), `$(shell command -v ...)` apparently works as expected. This seems to be some kind of optimization that is in effect whenever `SHELL` has the default value `/bin/sh` (observed by @chelmuth).

I adjusted the commit with the fixup 0af89bae16b8a5d9b1b33932c2c0a86c07614c8f.

chelmuth, 2021-11-09:
GNU Make fixed the bug in 2017 by https://git.savannah.gnu.org/cgit/make.git/commit/?id=1af314465e5dfe3e8baa839a32a72e83c04f26ef

commit f3a78774 (2021-11-09): Remove dependency from 'which' utility

Debian declared the 'which' utility as deprecated.

Fixes #4319
commit 9e485eb7 (2021-12-15): run: remove residual use of 'which' command

This is a follow-up commit for "Remove dependency from 'which' utility".

Thanks to John Karcher for reporting.

Fixes #4319
closed 2021-11-30 by chelmuth