genodelabs/genode #4320: base-hw: make run/* fails if `src/[base_src]` is imported from depot and `ld.lib.so` is in boot modules

jschlatow, 2021-11-09:
Although ld.lib.so is always required and included, it is still mentioned in some run scripts as a relict. However, I discovered a corner-case today in which some run scripts fail on copying ld.lib.so to the run dir. More precisely, it occurs for base-hw if the run script contains the following:

```
import_from_depot [depot_user]/src/[base_src]
[...]
build_boot_image "core init ld.lib.so ..."
```

Since ld.lib.so is already imported from the depot, it is present in the run dir when build_boot_image is called. The latter calls `run_boot_dir` in `tool/run/boot_dir/hw`, which skips building and copying ld-hw.lib.so if ld.lib.so is already in the run dir. However, if `ld.lib.so` is mentioned to be included in the boot image, the run tool tries copying the potentially non-existing bin/ld-hw.lib.so to the run dir.

chelmuth, 2021-11-10:
On second thought, I think it would be better to fix the faulty run scripts. The run tool cannot decide if the mentioned *ld.lib.so* is missing from *bin/* because it should be included from the depot or the script failed to produce the module in other ways, e.g., due to missing `build lib/ld/hw/`, which should lead to an abort.

jschlatow, 2021-11-10:
Actually, the run tool does decide whether the ld.lib.so is already present in the run dir (i.e. imported from depot) in line 51. If not, lib/ld/hw is built and ld.lib.so is copied from bin/ to the run dir (line 58). If lib/ld/hw is missing, the build will fail and lead to an abort.

In any way, at least for base-hw, there is no need to explicitly mention ld.lib.so in the run script. I think we should double-check whether this is also the case for other kernels and if not mirror the same behaviour before we can remove any mentioning of ld.lib.so from the run scripts. Nevertheless, I would still vote for including the proposed sanity check in order to prevent "shooting ourselves in the foot".

jschlatow, 2021-11-10:
@chelmuth, I checked the other kernels and only nova showed the same behaviour as hw. Commit 9fbb12a adds the same logic to the other kernels so that any mentioning of ld.lib.so can be removed from the run scripts.

jschlatow, 2021-11-23:
@chelmuth, as discussed today, commit 7aae819 would brake the use case in which someone wants to explicitly replace the ld.lib.so of the package with the one from bin/.

Apart from this, I still consider commit 9fbb12a a bugfix as it mirrors the same logic from hw and nova to all kernel variants, i.e. "copy ld.lib.so to run dir automatically if it was built automatically". Moreover, as this allows removing any mentioning of ld.lib.so from run scripts, it prevents copy-paste errors as described in my initial post.

Shall I make an effort in adapting the existing run scripts accordingly?

nfeske, 2021-11-23:
> Apart from this, I still consider commit 9fbb12a a bugfix as it mirrors the same logic from hw and nova to all kernel variants

I'm fankly not sure about it. NOVA and base-hw used to be exceptions, not the norm. I'm not quite sure that I really want to reverse that right now, mostly because I sense that we exchange one problem for another. Please correct me if I'm wrong, but with commit https://github.com/genodelabs/genode/commit/9fbb12aee17666dd9db2c15381ab1e4897d4271b, an ld.lib.so binary will always be taken from the bin/ directory if not specified in the boot_modules list. So if I'm using a ld.lib.so from a depot archive (like base-okl4 via import_from_depot) but I also happen to have a stale build of ld-okl4.lib.so rotting around in bin/, the run tool would overwrite the (more current) version from the depot with the stale version from bin/, wouldn't it?

I definitely acknowledge the inconvenience (and risks) of manually tweaking the build_boot_image list. But I think that the removal of ld.lib.so alone is not a big win. However, I have an idea in the back of my head how we can improve the handling of the boot module list in a more profound way: by tracking the build artifacts created during the `build` command. If you like, I can outline the idea. But I am admittedly a bit afraid of the distraction that this topic could cause us.

jschlatow, 2021-11-23:
> I'm fankly not sure about it. NOVA and base-hw used to be exceptions, not the norm. I'm not quite sure that I really want to reverse that right now, mostly because I sense that we exchange one problem for another. Please correct me if I'm wrong, but with commit [9fbb12a](https://github.com/genodelabs/genode/commit/9fbb12aee17666dd9db2c15381ab1e4897d4271b), an ld.lib.so binary will always be taken from the bin/ directory if not specified in the boot_modules list. So if I'm using a ld.lib.so from a depot archive (like base-okl4 via import_from_depot) but I also happen to have a stale build of ld-okl4.lib.so rotting around in bin/, the run tool would overwrite the (more current) version from the depot with the stale version from bin/, wouldn't it?

No, that's not the case. The run tool already has the logic that it builds `ld-*.lib.so` only if it is not already present in the run dir, i.e. copied from depot. Therefore, I think it is only logical to also copy the file from `bin/` in this case. From run script perspective, the current situation is that we never mention to build ld (because this is taken care of by the run tool) but still need to add `ld.lib.so` to the `build_boot_image` list...unless we use hw or nova.
On the contrary, if I modify an existing run script (which has `ld.lib.so` in `build_boot_image`) and add an `import_from_depot [depot_user]/src/[base_src]`, I must not forget to remove `ld.lib.so` from `build_boot_image`, because this will cause  the copying of a stale version from `bin/` to the run dir that you mentioned.

> I definitely acknowledge the inconvenience (and risks) of manually tweaking the build_boot_image list. But I think that the removal of ld.lib.so alone is not a big win. However, I have an idea in [...]

I agree that this is not a high-priority topic. Neither is the proposed change a big win...merely a matter of consistency.

nfeske, 2021-11-24:
Thanks for taking the time to debunk my concerns. You are very convincing. I'm really just a bit risk-averse when it comes to such integration issues because they hurt so much when going wrong. Shall we merge your commit to staging just after the release? This way, we can time-test the change during the next release cycle.

To clarify the bit about your observation that ld.lib.so is never manually specified in the build command, this is only because ld.lib.so is a dependency of all regular Genode components. As soon as such a component, e.g. init, appears in the list, ld.lib.so is built implicitly. It is just not explicitly mentioned. In a situation where no regular Genode component is built, e.g., taking all components from the depot except a POSIX app (w/o depending on ld.lib.so) that is built via the `build` command, one needs to put /lib/ld to the build arguments whenever one wishes to instrument the base library.

jschlatow, 2021-11-24:
> Thanks for taking the time to debunk my concerns. You are very convincing. I'm really just a bit risk-averse when it comes to such integration issues because they hurt so much when going wrong. [...]

That's a good plan.

tworaz, 2022-02-16:
@nfeske This commit series breaks setups in which default KERNEL variable value is set in etc/build.conf. I generally don't use other kernels besides hw much, so I always uncomment the KERNEL ?= hw line in the default build.conf file created by create_builddir script. The way it used to work was I could skip setting KERNEL=hw on make command line. With current master branch (8b7067d28999dac591f023181d8d43c904f1ad1d) this no longer works.

For example if I set KERNEL?=hw in build conf and run `make -C build/arm_v8a BOARD=virt_qemu run/log`, I get:
```
make[1]: Leaving directory '/home/tworaz/devel/genode/build/arm_v8a'
genode build completed
using 'ld-hw.lib.so' as 'ld.lib.so'
error copying "bin/ld-hw.lib.so": no such file or directory
    while executing
"file copy -force $src $dst"
    (procedure "copy_file" line 2)
    invoked from within
"copy_file bin/[kernel_specific_binary $binary] [run_dir]/genode/$binary"
    (procedure "copy_genode_binaries_to_run_dir" line 4)
    invoked from within
"copy_genode_binaries_to_run_dir $modules"
    (procedure "run_boot_dir" line 25)
    invoked from within
"run_boot_dir $binaries"
    (procedure "build_boot_image" line 35)
    invoked from within
"build_boot_image "core ld.lib.so init test-log""
    (file "/home/tworaz/devel/genode/repos/base/run/log.run" line 23)
    invoked from within
"source $include_name"
    ("foreach" body line 6)
    invoked from within
"foreach include_name [get_cmd_arg --include ""] {
	# first check if the include name is absolute
	if {[string first "/" $include_name] == 0} {
		puts ..."
    (file "/home/tworaz/devel/genode/tool/run/run" line 1115)
gmake: *** [Makefile:405: run/log] Error 1
gmake: Leaving directory '/home/tworaz/devel/genode/build/arm_v8a'
```

If I run `make -C build/arm_v8a KERNEL=hw BOARD=virt_qemu run/log` everything builds and runs as expected. Is this new behavior expected?

nfeske, 2022-02-16:
Thanks for reporting @tworaz. This is not intended.

I assume that the `?=` assignment is weaker than specifying `KERNEL` at the command line. The latter results the setting of `KERNEL` as environment variable, which is then evaluated in base/lib/mk/ld.mk. The weaker `?=` presumably does not have the same effect. As a quick test, could you try adding an `export KERNEL` to tool/builddir/build.mk?

tworaz, 2022-02-16:
@nfeske its not only `?=` that does not work. The same problem happens if I use `:=`.

Adding `export KERNEL` in tool/builddir/build.mk does fix the problem.

nfeske, 2022-02-16:
Thanks for testing. I have added the fix https://github.com/genodelabs/genode/commit/6dc9fa9e922a65a4831a459e6cb7768a42537b79 to staging now.

nfeske, 2022-03-02:
I have currently no further plans in this area. With the changes in Genode 22.02 - in particular the triggering of the ld.lib.so build at the build stage - I hope to have cleared up the situation. With the adoption of the `build_boot_image [build_artifacts]` pattern, the confusion will hopefully become less and less with the time.

@jschlatow What do you think?

jschlatow, 2022-03-02:
I'd consider this fixed by Genode 22.02.

closed 2022-03-02 by jschlatow