genodelabs/genode #4326: rom_filter: optionally skip top-level XML node from input ROMs when copying

jschlatow, 2021-11-16:
The rom_filter already supports copying an entire input ROM to the output ROM. Yet, since the top-level node of the output ROM is always generated by the rom_filter, there is no way to pass-through the content of an input ROM without wrapping it in an addition XML node.

A scenario where this is useful is, e.g., when we want to assemble the config of the platform driver (consisting of a list of '<device>' nodes) from multiple sources (e.g. a static part and a dynamic part depending on the current FPGA configuration).

jschlatow, 2021-11-16:
Commit b5e8d3e adds an argument `skip_toplevel`. If the argument evaluates to true, the top-level node of the input ROM is not copied but only its content.

nfeske, 2021-11-18:
I first wondered, may this use case already be covered by the `<input>` selection? The readme says:

> The '<input>' node may contain a hierarchy of '<node type="type">' nodes that
> denote a path within the XML input. If specified, the optional attributes
> 'attribute' and 'value' constrain the selection of the input depending of the
> presence and the value of the specified attribute.

So in principle, one could the skip the outer node by specifying the sub node that one wishes to copy.

However, this works only when copying only a single node. Given your list-of-devices example, I guess you want to copy *multiple* nodes, which cannot by expressed as an `<input>` declaration currently.

Your solution overcomes this limitation quite nicely! I merged your commit to staging.

nfeske, 2021-11-30:
The commit entered the master branch.

closed 2021-11-30 by nfeske