genodelabs/genode #4329: Enable support for etnaviv driver in the graphics stack

cnuke, 2021-11-18:
The following commits bring the already existing etnaviv support into a functioning state where 3D-acceleration is now supported on selected iMX8MQ-based systems, namely EVK and MNT Reform 2.

cnuke, 2021-11-18:
Commits d91bb9b, 3d5e44d, cb8bba0, 8e4a669, 8bf978d and 00a03e5 provide the necessary changes.

cnuke, 2021-11-18:
To be usable on Sculpt commit e168eb9 is also necessary to allow the `imx8mq_gpu_drv` access to the *platform* session. That however leads to the awkward situation where in the resource selection screen for the client component there are two possible choices for the *gpu* session, namely the one provided by the driver and the session potentially provided by the drivers subsystems (as the case on x86).

chelmuth, 2021-11-22:
I merged the series to staging. Before, I voiced my unease about the binary interface in 66bfb72 during an offline discussion. @cnuke and me agreed that a prettied up binary representation (as in [info_intel.h](https://github.com/genodelabs/genode/blob/abc9a2f232344dd4322c4529ee733e2f5d83c699/repos/os/include/gpu/info_intel.h)) isn't the ultimate solution. A better approach would be to use an XML representation like the new platform_drv implements for the info ROM. This may be considered when revising the GPU session.

cnuke, 2021-11-25:
@chelmuth could you please merge 59d4634 to staging (this fixes a page-fault @skalk encountered while testing the driver).

chelmuth, 2021-11-25:
Merged to staging.

commit 0ddeb4e9 (2021-11-22): libdrm: introduce Gpu session for etnaviv

* The Gpu session back end translates all DRM API requests of the
  client into matching Gpu session operations.

* Enable ioctl for etnaviv

Fixes #4329.
closed 2021-11-30 by chelmuth