genodelabs/genode #4330: platform_drv(arm): improve reconfiguration support

jschlatow, 2021-11-18:
When working on the Zynq SoC with a dynamically changing platform (due to FPGA reconfigurations), I noticed that the platform_drv doesn't handle this scenario properly at the moment:

If a device driver holds a Platform_session and its corresponding devices when the platform_drv is reconfigured, and if the corresponding policy disappear from the platform_drv's config, it will close the session of the device driver. The situation first appeared to be handled correctly, but leads to a page fault when the devices and the policy reappear.

jschlatow, 2021-11-18:
I narrowed this down to the fact that on a config update the Session_component is tried to be destroyed after the device model was already updated. This has the effect that the Device_components corresponding to the Session_component are not properly released and unregistered from the Rpc_entrypoint. Actually, the Device_component's destructor produces an error message should the Session_component still exist at the point of its destruction.

Commit a8e136b changes the destruction order. Unfortunately, fixing this causes the platform driver to exceed its cap quota after a few iterations of adding/removing devices from its config.

skalk, 2021-11-19:
@jschlatow Thanks for figuring out these problems. Yesterday @cnuke and me debugged the very same problem with the missing destruction of the device component and the capability leakage when restarting the GPU driver on ARM.
Concurrently I'm working on a inner re-design of the platform driver implementation, because in case of dynamically parsing and appearing PCI devices in the device namespace of the platform driver, the number of devices for the "virtual bus" of a client is not necessarily known in advance. Therefore the Device_component objects cannot get constructed at session creation time any longer. Moreover, this new dynamic requirements will change the semantic in a way that you can always have a platform session, potentially with no devices at all. But the platform driver can provide or close device objects dynamically within the platform session.

Regarding the page fault you observe, I'll try to reproduce it.

jschlatow, 2021-11-19:
@skalk thanks for the heads-up. Having a platform session with changing devices will become pretty useful for FPGA-reconfiguration scenarios on Zynq.

Re. reproducing the page fault: I just tried reproducing this on zynq_qemu in a much simpler setup using the network driver but failed to do so. In the scenario where the page fault occurs, I was using the ad9361 driver which accesses several devices. Some of the devices even persist between changes of the platform_drv's config.

skalk, 2021-11-19:
@jschlatow I've reimplemented your fixes slightly different in commit ids 5b46593 and dce8e75. Would you please try them? Do you still see the page-fault?

jschlatow, 2021-11-19:
@skalk with your commits I already got a page fault on the first destruction of the Session_component. Yet, if I ensure that `release_device()` is only called with a valid cap, I get the original behaviour (i.e. page fault on the second destruction of the Session_component).

jschlatow, 2021-11-23:
As it turned out, the page fault was caused by a rogue DMA controller that still performed transfers to the already freed DMA buffers. After correcting this, the changes to the platform_drv solve my use case nicely.

skalk, 2021-11-24:
With commit id 8a650ce and c69fa91 the page-faults and leaks in the current platform driver for ARM are solved. Therefore, I mark this issue as fixed, although all sessions with changing devices and policy still will get closed. The re-design to have permanent platform-sessions with appearing and disappearing devices I would like to address in a dedicated issue.

skalk, 2021-12-02:
Okay, as the issue is still open anyway, and I have now solved the re-configuration puzzle with commits d35fe40 and 5b256ca, we can just as well close the issue with those commits.
@jschlatow you might give them a try too solve your issues with fpga re-configuration.
@nfeske @chelmuth in my eyes commits d35fe40 and 5b256ca are ready to be reviewed.

jschlatow, 2021-12-03:
@skalk Splendid! I'll give it a try next week.

nfeske, 2021-12-07:
Thank you @skalk. I merge both commits to staging. While skimming over them, two minor details caught my eye.

In d35fe40deb665a4e7602d5b4c00208ab3c6e32f7, the `Owner` struct could probably be implemented without a `void *` member by making it `Noncopyable` and taking its own `this` pointer as token for its identity.

In 5b256caf2cbea2e0fd5f8af0e35e4d0ab9e3e13e, a few coding-style inconsistencies slipped in, like braces around single statement (https://github.com/genodelabs/genode/commit/5b256caf2cbea2e0fd5f8af0e35e4d0ab9e3e13e#diff-5072c8fc8b87b2a2fc29dba78b41b80179fe05085b6f321f0eafb62908f80324R193), and indentation/alignment (https://github.com/genodelabs/genode/commit/5b256caf2cbea2e0fd5f8af0e35e4d0ab9e3e13e#diff-5072c8fc8b87b2a2fc29dba78b41b80179fe05085b6f321f0eafb62908f80324R82). Nothing to get nervous about but I wouldn't mind a fixup.

skalk, 2021-12-07:
@nfeske sorry for the coding-style inconsistencies, I've prepared a fixup commit for it, see 3819183.

Regarding the proposal related to the `Owner` struct: I need to check whether a valid `Owner` was set for a `Device`.  An `Owner` is a `Session_component` or not yet existing.  If `Owner` object would be identical to its address (this pointer), I would again need to store its reference/pointer, and not the object itself within the `Device`. However, the whole construction should defend against storing the pure pointer to not risk to de-reference it by accident, but also identify the "owning" session apart from its potentially ambigous session label.

nfeske, 2021-12-07:
> An Owner is a Session_component or not yet existing.

Thank you for clarifying. That's what I was missing. I'm a bit embarrassed because this was actually a point we discussed in person. ;-)

Thanks also for the coding-style fixup, which I cherry-picked right away.

commit b2ed9bf4 (2021-12-07): autopilot: automated stress test for platform_drv

Fix #4330
commit f11d2b5b (2021-12-17): autopilot: automated stress test for platform_drv

Fix #4330
closed 2021-12-17 by nfeske