genodelabs/genode #4331: blit_helper: MMU fault on 64bit arm (rpi3)

tomga, 2021-11-19:
When investigating problem of MMU fault of framebuffer driver on rpi3 which I mentioned in #4303 I found that the problem is in function `copy_block_32byte` in [arm_64/blit_helper.h](/genodelabs/genode/blob/staging/repos/os/src/lib/blit/spec/arm_64/blit_helper.h).

Problem occurs if buffer addresses passed to this function are not aligned at 64bits.

I temporarily applied locally 9f626baad25eb735c6b00c3bb8aea34977bc342a which works but probably slower.

I did not create final fix as I'm not entirely sure if this function is supposed to support buffer addresses aligned to 32bits. If not then probably code in [blit.cc](/genodelabs/genode/blob/abc9a2f232344dd4322c4529ee733e2f5d83c699/repos/os/src/lib/blit/blit.cc#L40) should be changed (where comment suggests that only 32bit alignment is assumed).

chelmuth, 2021-11-22:
> I did not create final fix as I'm not entirely sure if this function is supposed to support buffer addresses aligned to 32bits. If not then probably code in [...]

Replacing *32-bit alignment* by *natural alignment* in the blitting library makes sense to me.

nfeske, 2021-11-22:
> Replacing 32-bit alignment by natural alignment in the blitting library makes sense to me.

I don't agree. Ideally, the blit library should not impose any alignment constraint to its users. Like memcpy, it should accept any any parameters at byte granularity. This way, it could be used to operate on 8bpp buffers (like an alpha channel).

It was originally created as a pragmatic utility to sharing some pixel-pushing code between applications (using 16bpp) long ago. By now, the library is in need of a profound overhaul that should consider both, weaker alignment constraints for the user, and better use of SIMD instructions whenever the parameters allow. Its implementation should best be tuned per architecture and blessed by real-world performance testing.

This, however, is a quite large topic, which I would not like to rush. It deserves much time and attention, including the questioning of the current API (once we are at it). So it is probably a topic we should plan for sometime next year.

For the immediate issue of the blit library in the rpi framebuffer, the problem may be worked around by truncating (x0) respectively rounding up (x1) the to-be-copied rectangular area to a multiply of two pixels. Since we are using 32bpp and the start of the buffer can be assumed to be 64-aligned, the rounding of the x coordinates would avoid the crash.

nfeske, 2021-11-22:
Alternatively, one could investigate why the misaligned access causes problems on the rpi3 whereas it is apparently works fine on the other 64-bit ARM platforms. 

May a different CPU setting be involved? (I vaguely remember that support for unaligned access can be granted to the user land)

Or are the page-table flags an issue - like the mapping of the framebuffer as uncached memory? If so, could the framebuffer be allocated as cached memory instead while adding explicit calls of `cache_clean_invalidate_data` for the touched areas?

chelmuth, 2021-11-22:
> > Replacing 32-bit alignment by natural alignment in the blitting library makes sense to me.
> 
> I don't agree. Ideally, the blit library should not impose any alignment constraint to its users. Like memcpy, it should accept any parameters at byte granularity. This way, it could be used to operate on 8bpp buffers (like an alpha channel).

This is what I meant by my comment: *blit.cc* should ensure natural alignment on the pointers before calling blit helpers. Nevertheless, I'll leave the issue to the experts.

nfeske, 2021-11-22:
> This is what I meant by my comment: blit.cc should ensure natural alignment on the pointers before calling blit helpers. 

Oh, that makes sense. I'm sorry and I agree with you.

tomga, 2021-11-28:
I'm not sure about what to do further with this issue. I have a workaround that makes the issue not critical for me (commit mentioned in initial report). I think I understand ideas behind proposed changes. But I don't feel I'm an expert in *blit* internals so possibly I'm not the best person to attempt to fix the problem. So will someone with greater experience in this part of code handle this issue at sometime later or would you like me to attempt to prepare a fix?

If I understand correctly I'd start by moving current `blit.cc` to 32bit spec subdirectory and create a new version for 64bit which would use 64bit variant helpers instead of 32bit variants. Is it the way to go with the fix?

skalk, 2022-01-10:
@tomga could you try whether commit 16e5d4a fixes the alignment error on the rpi3?

It is a fix for the mentioned problem only. The re-design @nfeske and @chelmuth have drafted is necessary IMO when looking at the blit library. However, it probably is beyond the solution of this specific issue.

tomga, 2022-01-10:
@skalk after applying build fixes commit a4f82b7f639b4f0b29b11966f98bd8d5f03e3d0b it works but given nature of fixes I applied I'm not sure that I found and used the same `aligned` function you wanted to use.

skalk, 2022-01-11:
@tomga I'm sorry, I did not want to produce you more work, but actually less. Unfortunatly, it seems I wasn't myself yesterday, and completely forgot to compile and test the code.

To take a step back: after thinking twice about the reason of the MMU alignment fault, I think we should change the rpi_fb_drv to use its own blit method. Thereby, we do not "pollute" the blit library with less optimized code. The problem with the rpi driver is that it gains its framebuffer as I/O memory, which is mapped as ordered device memory. Therefore, we see the alignment fault only here. Commit 89bd9dd sketches the idea. The former commit for the blit library can be discarded. @tomga would you git this commit a try? Do not worry, this time I've compiled it before ;-)

skalk, 2022-01-12:
@tomga I had to fix the semantic of the rpi-local blit function slightly. Commit 148fbb7 provides the solution now.

tomga, 2022-01-12:
@skalk I gave a try your patch and it causes:
```
no RM attachment (READ pf_addr=0x900000 pf_ip=0x1001b40 from pager_object: pd='init -> drivers -> fb_drv' thread='ep') 
Warning: page fault, pager_object: pd='init -> drivers -> fb_drv' thread='ep' ip=0x1001b40 fault-addr=0x900000 type=no-page
```
It is both on Rpi 1 and Rpi 3 - only addresses are different.

Leaving this implementation aside I'd like to ask you about:

> The problem with the rpi driver is that it gains its framebuffer as I/O memory, which is mapped as ordered device memory. Therefore, we see the alignment fault only here.

Do you think this may be easily changed and eliminate problem? I'll look more deeply into it probably.

In case it is not easy I'll probably look at copying more specialized `blit` implementation as if I correctly understand version for 32-bit arm should be much quicker than the one you implemented (yours copies always 16bit values). So it would hurt performance on all Rpi models even though the problem currently exists only on Rpi 3 in 64-bit mode.

tomga, 2022-01-13:
@skalk with this latest patch `demo` and `framebuffer` are working on Rpi 1 and 3.

Thanks for explanation of memory mapping. I think you are right that changes in mapping interface for this special case are not justified.

skalk, 2022-01-19:
@chelmuth @nfeske would you mind to merge commit 5c1e9d1 to staging? It solves this issue on rpi.

chelmuth, 2022-01-19:
Done.

commit 8ced7629 (2022-01-19): rpi_fb_drv: avoid alignment fault by own blit func

Fix #4331
commit 72489575 (2022-02-15): rpi_fb_drv: avoid alignment fault by own blit func

Fix #4331