genodelabs/genode #4333: cpu_balancer: abort due to `Out_of_ram` while applying config

nfeske, 2021-11-22:
On nova_x86_64, the cpu_balancer test fails with the following message (on Qemu):

```
[init -> cpu_balancer] affinity space=4x1
[init -> cpu_balancer] Warning: My label seems to be: 'init -> cpu_balancer'
[init] child "cpu_balancer" announces service "CPU"
[init -> cpu_balancer] Error: Uncaught exception of type 'Genode::Quota_guard<Genode::Ram_quota>::Limit_exceeded'
[init -> cpu_balancer] Warning: abort called - thread: ep
```

Since I suspected this issue be related to my recent work on #4324, I took a closer look but I'm not sure how to go about it.

The exception occurs while applying the configuration `Cpu::Config::apply` -> `Cpu::Session::config` -> `Cpu::Session::reconstruct` -> `Cpu::Session::construct` when trying to allocate from the session's `_md_alloc`.

The problem is that the allocation is caused by the CPU balancer, not the client, but is taken from the client's budget. If the budget is depleted, there is nothing we can do about it because the client is not involved in the reconfiguration.

I think it would be natural to account allocations that are taken on behalf of the CPU balancer (side effects of the reconfiguration) to the budget of the CPU balancer by passing a `Allocator &` into the `Cpu::Session::config`.

@alex-ab, what do you think?

alex-ab, 2021-11-23:
Thanks, for the report. Config updates were intentionally envisioned not to cause more memory consumption, but actually the implementation failed to adhere to the principle. I will take care to fix it asap. Using a limited allocator of the service itself, can bring down the service at the unpleasant moments, since the overall amount of clients and policy entries per client can't be estimated for sure on a dynamic system (as experienced by myself on Sculpt with the service).

nfeske, 2021-11-23:
> Using a limited allocator of the service itself, can bring down the service at the unpleasant moments [...]

It would definitely be the best to be able to avoid that, e.g., by pre-allocating all possible metadata during the thread creation when the client is prepared to pay for the costs of its new thread.

But should this not be possible, I think that allocation from the server's own budget during the reconfiguration can be justified. The reconfiguration is done on behalf of the (trusted) parent, not the (untrusted) client, after all. If the parent brings the server into trouble due to a configuration change, it should be the parent's responsibility to bail out the server (by providing a resource upgrade).

It would be very good if you find a way to let the clients pay for all the potential costs. But if this becomes too complicated or impossible (e.g., if the lifetime of the metadata must outlive the lifetime of the client), it would be fine to relax this principle and document it.

alex-ab, 2021-11-23:
With commit 30a3498b6596b89ed5075a2aea2c2f63a444475c and d4de2a005a24870afdf730ab27774029cece554f the memory consumption during config updates is avoided and policy meta data memory allocation happen solely during thread creation.

nfeske, 2021-11-23:
Thanks a lot @alex-ab for the rework! I merged your commits to staging.

Btw, I really like the sensible choice of names, like `update_if_active` or `reconstruct_if_active`. In commit https://github.com/alex-ab/genode/commit/d4de2a005a24870afdf730ab27774029cece554f, you removed the `const` from `Cpu::Session::report_state`. I'm not complaining but wonder why?

alex-ab, 2021-11-23:
Thanks. The change of using references (active_policy()) instead of pointers lead to complains about the constness, but i did not further investigate because i have to leave for today.

commit d124e265 (2021-11-23): cpu_balancer: avoid dynamic policy allocation

Pre-allocate all possible type of policy objects as part of the thread meta
state to avoid increased memory consumption due to different policy object
sizes. The cpu_balancer accounts the memory per client and can't forward
potentially occurring out-of-ram exceptions during config-ROM update phases.

Fixes #4333
closed 2021-11-30 by chelmuth