genodelabs/genode #4335: sandbox library to permissive and constraining

skalk, 2021-11-24:
While working on issue #4321 I've observed that a sub-init, whose start node has the 'managing_system' property set is passing that permission to its whole sub-tree. This is not really intended, because it gives e.g. firmware access to all drivers in a driver sub-init. A better solution would be that all init nodes on the path to core have to agree to set the 'managing_system' property, thereby being able to constrain this right exactly to one driver aka the platform driver.

While developing a solution, I could see that the 'phys_start'' and 'phys_size' constraints are set as heuristic within the sandbox library. This constraint for physical RAM allocation is also set for all sub-nodes of a 'managing_system' enabled node (formerly 'constrained_phys'). Although the code intended to parse existing phys_start and phys_size arguments and only set them when unavailable, it actually always sets them to the default values, see https://github.com/genodelabs/genode/blob/master/repos/os/src/lib/sandbox/child.cc#L666 for reference.

I would propose to only handle the 'managing_system' argument within the PD session construction and sandbox library. To always set it on its way through different child hierarchies, either true or false dependent on the start node properties. And finally do the phys_start/phys_size heuristic settings within core (dependent on the managing_system property), which are currently always the same values anyway.

nfeske, 2021-11-25:
Thanks for raising this issue.

I like the simplification with regard to the phys range but I found myself struggling with following the logic of the change in sandbox/child.cc.
The changed comment says "Unset the 'managing_system' argument unless explicitly permitted by the child configuration", and the code does:

```
  bool arg = (!_managing_system) ? false
           : Arg_string::find_arg(args, "managing_system").bool_value(true);
             Arg_string::set_arg(args, args_len, "managing_system", arg);
```

At first, I thought that it'd be a mere matter of style and tried make it easier to follow.

```
bool client_arg = Arg_string::find_arg(args, "managing_system").bool_value(false);
bool permitted  = (_managing_system && client_arg);

if (!permitted)
   Arg_string::remove_arg(args, "managing_system");
```

While doing this change, I figured that the default value of `bool_value` should better be false, not true. Otherwise, wouldn't the result be a permissive of no "managing_system" client argument is present?

When trying out the new version with demo.run and logging the PD session args in core's pd_root.h, I noticed that the `managing_system` attribute is actually never set. I expected this argument to be present for the platform driver's PD session. But it isn't.

The culprit lies in the fact that `managing_system=yes` is actually never specified as session argument for any PD (I grep'ed for it). To solve this problem, I figured that the sandbox library must distinguish PD sessions created by itself (for its immediate children) from PD sessions requested by a subsystem. For sessions initiated by itself, the sandbox library should imprint `managing_system=yes` to the session arguments if declared in the `<start>` node of the child.

I reworked the patch in this respect in the fixup 92796137b4baeffd65cf08f3e79e3a4b1b1db08f. I tested it with the demo.run scenario on base-nova and could observe that only the drivers init and the platform driver receives the `managing_system` permission at core.

Can you please check that the new version is still in line with your intentions?

skalk, 2021-11-25:
> Thanks for raising this issue.
>
> I like the simplification with regard to the phys range but I found myself struggling with following the logic of the change in sandbox/child.cc. The changed comment [...]

Following my original commit: it is not always permissive, because when `_managing` is false then I did not parsed the argument at all but set it to false. Thereby, it got always set, even when not set before, and this original commit worked as intended when you have init and a sub-init with all combinations of managing_system having been set or not, which I've tested yesterday.

However, you are right when the node (with _managing=yes) opens another PD session, e.g. a component not being init and not using the sandbox library, which is not setting the 'managing_system' property, the property would be set to true again. Therefore, your alternative approach is probably the better solution.

> 
> When trying out the new version with demo.run and logging the PD session args in core's pd_root.h, I noticed that the `managing_system` attribute is actually never set. I expected this argument to be present for the platform driver's PD session. But it isn't.
> 
> The culprit lies in the fact that `managing_system=yes` is actually never specified as session argument for any PD (I grep'ed for it). To solve this problem, I figured that the sandbox library must distinguish PD sessions created by itself (for its immediate children) from PD sessions requested by a subsystem. For sessions initiated by itself, the sandbox library should imprint `managing_system=yes` to the session arguments if declared in the `<start>` node of the child.
> 
> I reworked the patch in this respect in the fixup [9279613](https://github.com/genodelabs/genode/commit/92796137b4baeffd65cf08f3e79e3a4b1b1db08f). I tested it with the demo.run scenario on base-nova and could observe that only the drivers init and the platform driver receives the `managing_system` permission at core.
> 
> Can you please check that the new version is still in line with your intentions?

From reading the code, it is doing what I would expect, and I have tried it out with the actual scenario successfully.

commit 226e3f8b (2021-11-25): base: be more restrictive with 'managing_system'

* Only give managing_system permission when all parent nodes of the
  corresponding component agree in doing so.
* Move the physical memory constrains heuristic from sandbox library to core

Fix #4335
closed 2021-11-30 by chelmuth