genodelabs/genode #4336: nova: determine more accurate TSC frequency for Skylake and newer CPUs

cproc, 2021-11-25:
While printing log messages with trace timestamps converted to milliseconds by dividing with the TSC frequency determined by NOVA, I noticed a slight drift of the time values compared to the timestamps printed by the Linux 'ts' tool. It turned out that the determined TSC frequency (2500000 kHz) was not exactly accurate for the CPU in use (Intel i5-6300U) and that there exists a new method to determine the TSC frequency more accurately for Skylake and newer CPUs.

From the Intel manual chapter "Determining the Processor Base Frequency":

```
For Intel processors in which the nominal core crystal clock frequency is enumerated in CPUID.15H.ECX and the core crystal clock ratio is encoded in CPUID.15H (see Table 3-8 "Information Returned by CPUID Instruction"), the nominal TSC frequency can be determined by using the following equation:

Nominal TSC frequency = ( CPUID.15H.ECX[31:0] * CPUID.15H.EBX[31:0] ) ÷ CPUID.15H.EAX[31:0]

For Intel processors in which CPUID.15H.EBX[31:0] ÷ CPUID.0x15.EAX[31:0] is enumerated but CPUID.15H.ECX is not enumerated, Table 18-68 can be used to look up the nominal core crystal clock frequency.

Table 18-68.  Nominal Core Crystal Clock Frequency

Processor Families/Processor Number Series     Nominal Core Crystal Clock Frequency

6th and 7th generation Intel® Core processors (does not include Intel® Xeon® processors) 24 MHz
Next Generation Intel® Atom processors based on Goldmont Microarchitecture with CPUID signature 06_5CH (does not include Intel Xeon processors) 19.2 MHz

NOTES:
1. For any processor in which CPUID.15H is enumerated and MSR_PLATFORM_INFO[15:8] (which gives the scalable bus frequency) is available, a more accurate frequency can be obtained by using CPUID.15H.
```

I implemented this method in NOVA commit [1] and it determined a TSC frequency of 2496000 kHz for the i5-6300U CPU, which solved the log timestamp drift issue and might improve the accuracy of other TSC-based time calculations in Genode as well. Only tested with this specific CPU so far.

https://github.com/cproc/NOVA/commit/50e53d9825a124bc0fb96a363b6a05224314a2d3

alex-ab, 2022-01-18:
Thanks @cproc. I took your commit and adjusted and tested it with older machines. Please give 2a7a4e12be135509f2734d0b109ee0e5b7a6a03b a try.

alex-ab, 2022-01-18:
Sorry, 16d96c493a37da9338d0c8e373ad9588e06cd94b is the right commit.

cproc, 2022-01-19:
@alex-ab: thanks, works for me on the x260

commit 3a39cb86 (2022-01-19): nova: read out more accurate tsc frequency

on newer CPUs.

Fixes #4336
closed 2022-02-15 by nfeske