genodelabs/genode #4337: gdb_monitor "__cxa_pure_virtual called" error on 21.05 and 21.08

S7rizh, 2021-11-30:
On the latest versions of Genode (21.08, 21.05) gdb_monitor component fails on construction with the following error:

`[init -> gdb_monitor] Error: __cxa_pure_virtual called, return addr is 0x12c139`

I have tried to run different scenarios (gdb_monitor, gdb_monitor_interactive, or own run script) but all of them fail with an identical error. I suppose that this problem occurred after the update to the new toolchain. I also tried to figure out what is the cause of the problem and got the following stack trace using qemu gdb stub:

```
#0  __cxa_pure_virtual () at /genode_experiment/repos/base/src/lib/cxx/misc.cc:32
#1  0x000000000012c259 in Linker::Root_object::~Root_object()::{lambda(Linker::Dependency&)#1}::operator()(Linker::Dependency&) const (
    __closure=0x401fe6f8, d=...) at /genode_experiment/repos/base/src/lib/ldso/include/linker.h:361
#2  0x00000000001368c9 in Genode::Fifo<Linker::Dependency>::dequeue<Linker::Root_object::~Root_object()::{lambda(Linker::Dependency&)#1}>(Linker::Root_object::~Root_object()::{lambda(Linker::Dependency&)#1} const&) (
    this=0x1c61e0 <unmanaged_singleton<Linker::Binary, 8, Genode::Env&, Genode::Heap&, Linker::Config const&, char const*>(Genode::Env&, Genode::Heap&, Linker::Config const&, char const*&&)::object_space+1432>, func=...) at /genode_experiment/repos/base/include/util/fifo.h:168
#3  0x0000000000133c71 in Genode::Fifo<Linker::Dependency>::dequeue_all<Linker::Root_object::~Root_object()::{lambda(Linker::Dependency&)#1}>(Linker::Root_object::~Root_object()::{lambda(Linker::Dependency&)#1} const&) (
    this=0x1c61e0 <unmanaged_singleton<Linker::Binary, 8, Genode::Env&, Genode::Heap&, Linker::Config const&, char const*>(Genode::Env&, Genode::Heap&, Linker::Config const&, char const*&&)::object_space+1432>, func=...) at /genode_experiment/repos/base/include/util/fifo.h:182
#4  0x000000000012c2dc in Linker::Root_object::~Root_object (
    this=0x1c61e0 <unmanaged_singleton<Linker::Binary, 8, Genode::Env&, Genode::Heap&, Linker::Config const&, char const*>(Genode::Env&, Genode::Heap&, Linker::Config const&, char const*&&)::object_space+1432>, __in_chrg=<optimized out>)
    at /genode_experiment/repos/base/src/lib/ldso/include/linker.h:360
#5  0x000000000012fffa in Linker::Binary::Binary (
    this=0x1c5c48 <unmanaged_singleton<Linker::Binary, 8, Genode::Env&, Genode::Heap&, Linker::Config const&, char const*>(Genode::Env&, Genode::Heap&, Linker::Config const&, char const*&&)::object_space>, env=..., md_alloc=..., config=..., name=0x1446b2 "binary")
    at /genode_experiment/repos/base/src/lib/ldso/main.cc:368
#6  0x000000000013786e in Unmanaged_singleton_constructor::call<Linker::Binary, Genode::Env&, Genode::Heap&, Linker::Config const&, char const*&> (
    dst=0x1c5c48 <unmanaged_singleton<Linker::Binary, 8, Genode::Env&, Genode::Heap&, Linker::Config const&, char const*>(Genode::Env&, Genode::Heap&, Linker::Config const&, char const*&&)::object_space> "\230\321\025")
    at /genode_experiment/repos/base/src/include/base/internal/unmanaged_singleton.h:58
#7  0x0000000000135ce0 in unmanaged_singleton<Linker::Binary> ()
    at /genode_experiment/repos/base/src/include/base/internal/unmanaged_singleton.h:85
#8  0x0000000000132c55 in Component::construct (env=...) at /genode_experiment/repos/base/src/lib/ldso/main.cc:812
```

I wonder if this is a problem with the configuration of my build environment (currently I am building everything using Genode development containers) and what possible fixes to this problem are available

chelmuth, 2021-11-30:
Sorry but I can't reproduce what you posted as our nightly test of `make run/gdb_monitor KERNEL=nova` succeeds. Please see the attached [log file excerpt](https://github.com/genodelabs/genode/files/7626557/log.txt)

S7rizh, 2021-12-17:
I figured out what was the problem. It turned out that I was building the scenario with the `CC_OLEVEL=-O0` option and it resulted in this strange error. Seems that issue is resolved but still wondering what is the cause of such a behavior

closed 2021-12-17 by S7rizh

nfeske, 2021-12-17:
Thank you very much for sharing the outcome with us, and for updating/closing the issue. Should you find out more about the mystery regarding the use of -O0 in this scenario, I'd love to learn about it.

One effect that we repeatedly observed is the increased stack usage when using -O0. Since function calls are not inlined, each call nesting level consumes a stack frame. But if that would be the trouble, the symptom would probably be different (a page fault in the stack area).