genodelabs/genode #4338: platform_drv: introduce clocks, power, reset domains

skalk, 2021-12-06:
Beside the latest, generic platform driver for ARM systems there exist two derivates for allwinner and imx8mq, which share the need to configure clocks, reset- and power-domains. It would be nice to share the bureaucracy needed to hold information about device settings related to these. Although, those settings ae highly hardware-specific, it would be beneficial to share the data-structures. In the generic case, they would simply stay empty.
Moreover, when sharing these constructs, it is a good time to get rid of the Driver::Env within the platform driver.

tomga, 2021-12-06:
>  Moreover, when sharing these constructs, it is a good time to get rid of the Driver::Env within the platform driver.

@skalk, I'd like to ask you about this. In my rpi branch I used Driver::Env as a way to extend platform driver for adding rpi specific functions. It is in 61fe61ed9657984b72697266e64bffc6f1069e9b and 236c9b9c61cd0fa49c98938c19c7de2838301f96 for adding rpi `mbox` as a device locally managed and in 338a186ba2a0d6bf08368965248aca2d632e66e7 to add rpi specific rpc service discussed in #4303.

With removal of this Driver::Env I'll loose ability to cleanly extend generic platform driver. Could you propose some alternative for such extensions viable after your change?

skalk, 2021-12-06:
@tomga Uups, you have reminded me to change the rpi_platform_drv too, which I had forgotten. Fixup commit 84d5d19 changes the platform driver for Rpi accordingly. It also has to make use of the "mbox". Therefore, this implementation might be insightful for you?! However, I could not test it on hardware right now, because I'm in home-office today.
Does this implementation clarifies your question?

tomga, 2021-12-06:
@skalk, yes, thank you. I've seen the possibility to make a copy of `main.cc` but somehow felt that copying 50 lines of code to add one or two additional was not appropriate. I wouldn't propose this but I have no problem with this solution either. Especially that I don't see how to do it in different manner without polluting default implementation with some extension point.

skalk, 2021-12-07:
@tomga actually, I've reacted with that change to a wish of @nfeske, who liked to have the "main.cc" under control of the concrete driver implementation. Thereby, it is possible e.g., to add another role to the driver like being a pin control service in addition to the platform driver role etc..
Now, one has to duplicate some addittional lines of the original driver implementation. But I think the difference with respect to the former "env.h" file is neclectable.

tomga, 2021-12-07:
@skalk, maybe my comment was written too quickly as I did not take into account that you removed `env.h`. Earlier, when I looked for a way to add `mbox` device and service my choice was different: either copy `main.cc` or add few lines. Now you mostly moved code to a different place. Thanks for clarification.

skalk, 2021-12-07:
@tomga you're welcome!

nfeske, 2021-12-07:
Thank you for this very welcome consolidation work, Stefan! I merged your commit to staging.

commit 58923ce3 (2021-12-07): platform_drv: introduce clocks, power, reset

Share datastructures for clock, power and reset related configurations
per device. In the generic platform driver component these structures
are kept empty. Driver derivates can fill the clocks settings, power and
reset switches with life. The former Driver::Env gets removed.

Fix #4338
commit 6cc60479 (2021-12-17): platform_drv: introduce clocks, power, reset

Share datastructures for clock, power and reset related configurations
per device. In the generic platform driver component these structures
are kept empty. Driver derivates can fill the clocks settings, power and
reset switches with life. The former Driver::Env gets removed.

Fix #4338
closed 2021-12-17 by nfeske