genodelabs/genode #4340: net: fix tail size calculation in Ethernet_frame

jschlatow, 2021-12-08:
The const-variant of the `Ethernet_frame::data()` method contains an erroneous calculation of the tail size. This leads to the size guard throwing an exception when trying to parse TCP packets that only contain the TCP header.

The correct implementation of the method is found in the non-const variant.

nfeske, 2021-12-08:
Thank you @jschlatow for reporting and the fix. I merged your commit to staging.

I'm little curious about the symptoms that led you to this issue. Is there an anecdote to share?

m-stein, 2021-12-14:
@jschlatow Thanks for having investigated and fixed this one! As far as I can tell from looking at the code, you are absolutely right.

closed 2021-12-17 by nfeske