genodelabs/genode #4341: lwip: socket_fs plugin does not support large backlog numbers in listen file

jschlatow, 2021-12-09:
When running the iperf port from genode-world on Sculpt, I encountered a problem when using lwip instead of lxip. The former reported a failed call to `listen()`, hence I had a look. As it turns out, iperf calls `listen(sockfd, INT_MAX)`, yet the lwip plugin only uses a character buffer of length 8 to parse the number. This is too small for the 10-digit decimal number produced by INT_MAX.

chelmuth, 2021-12-14:
Good catch. Do you now why the code checks for `count < 11` with a buffer sized `buf[12]`?

Also: What's your opinion about the code copying `count + 1` bytes from the source buffer? Isn't this out-of-bounds access?

jschlatow, 2021-12-14:
I was wondering about the same thing but haven't looked into that. I believed the `count + 1` being related to something like a null termination that is not accounted by count since `count + 1` is consistently used at other places in the code. What is not consistent though is that, at some places in the code, count is checked for ` < sizeof(buf)` and at other places ` < sizeof(buf)-1`. I have no idea whether this is intentional though.