genodelabs/genode #4344: Eliminate warnings as much as possible

skalk, 2021-12-13:
During the resolution of the -Wconversion issue #23 I've stumbled again about other warnings, mostly from contrib code. The problem is obvious: the more warnings we see there is less attention given to them. Therefore, I would like to minimize the warnings of all build targets as much as possible.
This is more a long-term task, and of course everybody is welcome to participate. There is a rule of thumb I would propose. In Genode's own code we try to eliminate warnings completely. In contrib code: more cosmetical warnings about unused variables or functions etc. can simply be turned off per compilation unit or contrib library. Everything else should be investigated more deeply, whether it points to a real problem. If that question can be answered with yes, a patch should solve the warning, otherwise we make it silent.

nfeske, 2021-12-15:
Thanks a lot @skalk for the initiative. I merged your highly appreciated fixes to staging.

nfeske, 2023-08-24:
This issue has become stale. So I'm closing it.