genodelabs/genode #4345: base-hw: merge <spec>/kernel/interface_support.h into kernel/interface.h

chelmuth, 2021-12-14:
The incarnations of *interface_support.h* for arm, arm_64, riscv, and x86_64 could be unified to the following type definitions. These could be placed in *interface.h* and the support headers be removed.

```c++
/* Genode includes */
#include <base/stdint.h>

namespace Kernel {
	typedef Genode::umword_t Call_arg;
	typedef Genode::umword_t Call_ret;
	typedef Genode::uint64_t Call_ret_64;
}
```

@skalk @m-stein  Do you have any objections?

m-stein, 2021-12-14:
I just asked my self why there is the need for a ``Call_ret_64``. Will look this up. Apart from that I think this change is a good idea ;)

m-stein, 2021-12-14:
As far as I recall, it is valid to consider the support for 64 bit integers mandatory and, on most platforms, also efficient. Therefore, how about making ``Call_ret`` always 64 bit and getting rid of ``Call_ret_64`` together with ``call64(...)``?

chelmuth, 2021-12-20:
> As far as I recall, it is valid to consider the support for 64 bit integers mandatory and, on most platforms, also efficient. Therefore, how about making `Call_ret` always 64 bit and getting rid of `Call_ret_64` together with `call64(...)`?

I disagree with your reasoning and vote for a kernel ABI that uses one standard register for return values as long as there's no requirement for a predefined value type (like that for the time value). Additionally, I rather identified demand for enumerated result cases rather than magic negative values.

m-stein, 2021-12-20:
:+1: for enumerated result cases.

closed 2022-02-28 by chelmuth