genodelabs/genode #4349: Remove `Trace::Session::subject_info` RPC

nfeske, 2021-12-15:
This interface is superseded by the shared-memory-based `for_each_trace_subject` mechanism.

The old mechanism is a leftover from issue #3610. I stumbled upon it because I added a warning message to base-okl4 whenever the maximum message size is exceeded, which happens to happen when returning the `Subject_info` as message payload.

nfeske, 2021-12-16:
In the commit https://github.com/genodelabs/genode/commit/906e2d0b4b0174d9d9abc9d75b2746b4cfa88a88, I covered the trace logger and the tests exercised by the depot_autopilot.

Our buildbot points out that the following additional components still rely on the old interface:

* vfs_trace
* cpu_balancer
* trace_subject_reporter

I'll update the trace_subject reporter.

@alex-ab would you adjust the cpu_balancer?
@ssumpf can you take care of vfs_trace?

ssumpf, 2021-12-16:
@nfeske: sure

alex-ab, 2021-12-16:
@nfeske: yes

ssumpf, 2021-12-16:
@nfeske: 3847a0d takes care of `vfs_trace`

alex-ab, 2021-12-16:
@nfeske: 8d95978c20495d29acc819212598a31062e059b4 adjust the cpu_balancer

nfeske, 2021-12-16:
Thanks @alex-ab and @ssumpf for responding so quickly.

chelmuth, 2021-12-16:
`make -C build/x86_64 run/migrate KERNEL=nova` fails like follows.

```
.../repos/base/src/test/migrate/main.cc: In member function ‘void Migrate::check_traces()’:
.../repos/base/src/test/migrate/main.cc:119:43: error: ‘struct Genode::Trace::Connection’ has no member named ‘subject_info’
  119 |    Trace::Subject_info const info = trace.subject_info(trace_id);
      |                                           ^~~~~~~~~~~~
.../repos/base/mk/generic.mk:61: recipe for target 'main.o' failed
make[3]: *** [main.o] Error 1
var/libdeps:185: recipe for target 'test-migrate.prg' failed
make[2]: *** [test-migrate.prg] Error 2
make[2]: *** Waiting for unfinished jobs....
```

alex-ab, 2021-12-16:
0256a755713a06439dc957df97e994e8dfb35cfe fixes the test/migrate

closed 2021-12-17 by nfeske