genodelabs/genode #4351: riscv: failed tests in depot_autopilot

chelmuth, 2021-12-17:
riscv_qemu made its first rounds in the nightly depot_autopilot runs and did very well. Nevertheless, there the following remaining issues.

- [ ] test-libc fails on
  ```
  [init -> depot_autopilot] 1.969 [init -> test-libc] Malloc: check really large allocation
  Kernel: core -> entrypoint raised a fault, which should never happen ip=0xffffffc0000181e2 fault-addr=0xffffffd011541fff type=no-page
  ```
- [ ] test-rm_fault_no_nox times out after
  `[init -> test-rm_fault] test WRITE faults on read-only binary and read-only attached RAM`
- [ ] test-trace_logger times out
  Seems the trace messages of `trigger_once` are missing

chelmuth, 2021-12-20:
@ssumpf I was a bit puzzled about the following lines in the log (note the RAM-size values).

```
spawn qemu-system-riscv64 -nographic -serial mon:stdio -kernel var/run/depot_autopilot/boot/image.elf -m 128 -machine virt -cpu rv64,priv_spec=v1.10.0 -bios default
OpenSBI v0.5 (Oct  9 2019 12:03:04)
...
Genode 21.11-50-gd62d67de66 <local changes>
504 MiB RAM and 64533 caps assigned to init
```

And indeed, the following patch "fixes" *test-libc* (at least).

```diff
+++ b/board/riscv_qemu/qemu_args
@@ -1,2 +1,2 @@
--m 128 -machine virt -cpu rv64,priv_spec=v1.10.0
+-m 512 -machine virt -cpu rv64,priv_spec=v1.10.0
 -bios default
```

ssumpf, 2021-12-20:
> @ssumpf I was a bit puzzled about the following lines in the log (note the RAM-size values).
>
> ```
> spawn qemu-system-riscv64 -nographic -serial mon:stdio -kernel [...]

:+1:

nfeske, 2022-11-17:
Apparently, all tests on Qemu's virt platform for RISC-V are happy by now.

closed 2022-11-17 by nfeske