genodelabs/genode #4352: Implementing a trace_recorder component

jschlatow, 2021-12-17:
This issue is intended for keeping track of ideas and progress w.r.t. enhancing the tracing capabilities in genode.

The ideas written down below evolved from some experiments I conducted for extracting [CTF traces](https://diamon.org/ctf/) in Sculpt in order to record traces of component interactions. My first idea was implementing this as a vfs plugin, similar to vfs_trace. While this worked as expected and produced first results that I was able to conventiently visualise and analyse with TraceCompass, I recently identified a few limitations of this approach.

The vfs_trace plugin already provides convenient interactive access to the TRACE session via the vfs. This enables the use case of extracting debug output from a component without impacting its performance by the extensive use of the LOG session. The debug output can then be evaluated by accessing the raw trace buffer via vfs_trace.

When it comes to extracting binary traces such as the aforementioned CTF, we need a few more ingredients, though. First, as we usually want to minimise the performance penalty caused by the trace policy, the raw trace events found in the trace buffer are not necessarily in the intended target format. Moreover, depending on the traced time interval, the produced amount of data might easily exceed the size of the trace buffer, in which case the buffer will wrap around and old events are overwritten. In case of the CTF traces, my trace monitor therefore periodically samples the trace buffer and appends the new events into a file after adding some header information. Second, in order to trace component interactions, it is helpful to start/stop tracing of multiple components at the same time. This is better achieved by specifying the to-be-traced components, threads and inserted trace policies in a config ROM (just like the trace_logger) instead of doing this interactively via a vfs.

After working on the CTF traces and investigating a network performance issue, I further noticed that this approach of recording continuous binary traces is not limited to CTF, though. It would be very useful to use the very same mechanism for recording pcap traces as well. Furthermore, having a mechanism to simultaneously record CTF, pcap and debug traces would also be quite helpful.

I'm therefore going to implement a dedicated trace recorder component with a config interface and a simple GUI that takes care of the aforementioned trace policy and trace buffer management. When sampling the trace buffers, the trace recorder will push the trace events to an arbitrary number of registered backends. The backends will take care of writing the respective target format. I think it is best to start with built-in backends for ctf, log and pcap, yet this structure would also allow for a plugin interface if this became useful in the future.
The config ROM of this component would then look much like a trace_logger config:

```
<config period_ms="1000" default_buffer="1M">
   <policy label_suffix="nic_drv"    policy="ctf_rpc_log">
      <ctf/>
      <log/>
   </policy>
   <policy label_suffix="nic_router" policy="ctf_rpc_pcap" buffer="4M" thread="ep">
      <ctf/>
      <pcap/>
   </policy>
</config>
```

The noteworthy addition here is the mentioning of the writer backends for every policy. In the example above, let's assume the trace policy `ctf_rpc_log` writes raw log_output events as well as binary ctf events. We therefore want to forward all events to a ctf writer and a log writer, each of which formatting and writing the events it is interested in into the respective target file. In order to distinguish the different binary formats of trace events it will likely become necessary to introduce a type field for evaluation by the writer backends.

A GUI can be used for starting/stopping a trace recording as specified by the config ROM.

Of course, the vfs_trace plugin will still remain useful, yet the trace_recorder would supersede the trace_logger.

jschlatow, 2022-07-04:
fyi, I've been working on this issue once in a while and am currently using the trace_recorder for capturing multi-component traces to identify network bottlenecks.

The current implementation supports CTF and pcapng output. It can be found on my [trace_recorder](https://github.com/jschlatow/genode/tree/trace_recorder) branch. There's still some fine tuning going on. I'll let you know once I consider it ready for merging.

jschlatow, 2022-08-01:
@chelmuth I think the commit series are ready for a review so that we may consider it for the upcoming release. I referenced the corresponding commits. Since the usage is not quite self-explanatory, I plan to write a genodians article for more background information and usage examples. Please let me know if you have any feedback.

chelmuth, 2022-08-10:
I merged the entire series accompanied by some minor style fixes I discovered during the review to staging. Looking forward to your article.

jschlatow, 2022-08-10:
Thanks @chelmuth for the review and for merging the series. I'll provide the missing fixups (NIC router README and policy name) soon.

jschlatow, 2022-08-12:
@chelmuth I've pushed four fixup commits that address the missing README and correct the policy name(s): 8d52b3e, c175686, fd26369 and 8868db3

chelmuth, 2022-08-15:
Thanks, merged all to staging.

m-stein, 2022-08-16:
@chelmuth @jschlatow I added a minor fixup to the NIC router commit:
2685f7efd6 fixup "nic_router: add trace_packets config option" (README)

Furthermore, I would suggest to add a direction argument to ``trace_eth_packet`` (as with ``verbose_packets`` in the router) because, also in general for networking components, a packet is either received or sent and this information is often vital for debugging.

jschlatow, 2022-08-16:
@m-stein Thanks for the clarification in the README.

I also like your suggestion about the direction argument. I haven't considered this because there is no way to reflect this in the pcapng format, however, I recognise its usefulness.

jschlatow, 2022-08-16:
@chelmuth I added a fixup series that adds a direction argument to `trace_eth_packet()`: e2ba313, a8c0693, b8cfda7, c54b122, aec9601 and ef5c26e

nfeske, 2022-08-26:
While reviewing commits for the upcoming release, commit cf904e0a5d92fe7c5b467454d7b49870f615ad92 caught my eye. In my opinion, this API change is too lax.

The `New_file` class is strictly meant for cases where a new file is created. If a file with the same name exists, it gets replaced.

The use case of appending content to a file should better be modeled by a new type, e.g., `Appending_file`. A dedicated type is justified here because the distinction between "new" and "append" is not just a minor detail. A "new" file potentially deletes information. The content is created as a whole at once. There is no mutation of existing information. In contrast, when appending, no existing data get lost. But the origin of the information is not clear cut because the resulting content is a conglomerate of existing state and the append operation.

Aside the design consideration of using distinct types, let me highlight that `bool` arguments should better be avoided in public APIs because they obscure the caller side. It is better practice to introduce a new type, e.g., an enum class, or a `bool` wrapped in a `struct`for the argument instead. Also the use of default arguments is generally shunned from public APIs.

@jschlatow Would you consider creating a follow-up commit that implements the append feature as a new class?

jschlatow, 2022-08-29:
@nfeske Thanks for sharing your view on the API change. I completely agree with you. I am going to provide a follow-up commit.

jschlatow, 2022-08-29:
@nfeske please consider 08a5407 as the said follow-up commit. To prevent code duplication, I let `New_file` inherit from `Append_file`. Let me know if you see any problems with that.

nfeske, 2022-08-30:
Thank you for quick response. I'll review your commit as soon as the release is out.

nfeske, 2022-09-05:
> To prevent code duplication, I let New_file inherit from Append_file. Let me know if you see any problems with that.

That's perfectly fine. Thank you for implementing the change. I merged your commit to staging.

chelmuth, 2022-09-07:
Unfortunately, the following change breaks `make run/test PKG=test-fs_rom_update KERNEL=linux BOARD=linux` and `make run/intel_fb KERNEL=nova`.

https://github.com/genodelabs/genode/commit/870d243bcd5a9ba261cfaa3efe81290b8bc43dc6#diff-95986eddc1e977463efdbcf48e77f8f8454a04777ef84f3d8a86784b001cf562R718-R720

The reason is that the `seek` position is managed by the `Vfs_handle` user entirely and, thus, is not reset automatically at

https://github.com/genodelabs/genode/commit/870d243bcd5a9ba261cfaa3efe81290b8bc43dc6#diff-95986eddc1e977463efdbcf48e77f8f8454a04777ef84f3d8a86784b001cf562R833

Adding `_handle.seek(0)` fixes the issue but I'm unsure if it may be another step in the wrong direction of equalizing `New_file` and `Append_file`, which seem to have subtle differences. What's your opinion?

jschlatow, 2022-09-07:
AFAIAC, resetting the `seek` position in `New_file` makes perfect sense since the intention is to overwrite any existing data. It somehow escaped my attention that `seek` is not reset automatically by `ftruncate`.

nfeske, 2022-09-07:
I think that `New_file` should not do any seeking. It smells like the intertwining of `New_file` and `Append_file` is not natural.

The inheritance is merely on the account to avoid code duplication. @jschlatow would you be open to move the code shared between both classes to appropriately named functions instead of using inheritance? I remember a similar discussion with @cproc, where we ended up with the `with_raw_file_content` function (#4372) in the same spirit.

In our case, there could be a global `append` function with a signature as follows:

```
static inline Append_result append(Vfs::Vfs_handle &, Entrypoint &, char const *src, size_t size);
```

This code of `_init_handle` could be moved to a function named `vfs_handle_for_writing` without doing the seeking.

Both classes could then use these functions without having any relationship with each other.

jschlatow, 2022-09-08:
@nfeske I've stumbled upon an issue with implementing the function `vfs_handle_for_writing`. It requires a `Directory::Path &` argument, however, `Directory` has only been forward-declared when declaring `vfs_handle_for_writing`. I cannot come up with a pretty way to resolve this dependency. I'm thus thinking about implementing a `Writeable_file` class instead that hosts the handle initialisation and append code and from which `New_file` and `Append_file` can inherit. What do you think?

jschlatow, 2022-09-08:
I pushed the fixup commit 3e2525d, which implements the discussed changes.

nfeske, 2022-09-08:
That is very much appreciated. I like how the new version makes the behavior of both `New_file` and `Append_file` obvious at the first glance.

nfeske, 2022-09-08:
Both classes could be slightly simplified because there is no need to remember the `_alloc` and `_fs` references.

jschlatow, 2022-09-08:
@nfeske, @chelmuth I addressed your comments in e94a5f5.

nfeske, 2022-09-21:
In the meanwhile, the residual rework of the os/vfs.h API has reached the master branch.

closed 2022-09-21 by nfeske