genodelabs/genode #4355: CBE SB control: don't encrypt invalid keys

m-stein, 2021-12-19:
This issue is related to https://github.com/cbe-project/cbe/issues/72. Originally it was only meant for the CBE port update but I encountered also several problems with testing the new CBE in Genode. More specifically, the ``vfs_cbe`` test doesn't build with staging and even when fixing the build, there remain several problems at runtime.

nfeske, 2021-12-20:
> the vfs_cbe test doesn't build with staging

This comment makes me wonder because the nightly build does not report any build error. Also, the cbe_tester.run script is executed successfully (runs on base-linux).

m-stein, 2021-12-20:
The cbe_tester run script only tests the CBE core logic very directly whereas the ``vfs_cbe`` includes also the VFS plugins and VFS-based apps that are related to the CBE. As far as I can tell from my ``autopilot.list``, ``vfs_cbe`` is not tested nightly. It is part of the ``tool/cbe_autopilot`` however. I don't know whether ``cbe_autopilot`` is executed once in a week as was the original idea behind this tool.

Furthermore, I've found already two fixes with one of them being perfectly reasonable to me.

m-stein, 2021-12-20:
Unfortunately I cannot redo tests on a fresh repo's staging because ``gnu.org`` seems to be down for a while now.

m-stein, 2021-12-20:
(the tests require ports like ``bash`` and ``coreutils``)

m-stein, 2021-12-20:
I forgot to mention that ``cbe_tester`` succeeds with staging on my machine.

chelmuth, 2021-12-20:
I can confirm that *cbe_tester* succeeded at least since Friday. But, *tool/cbe_autopilot* is currently not enabled in *ci-tools*.

m-stein, 2021-12-20:
We could transfer the CBE autopilot tests to run scripts and add them to the autopilot list as well. Reasons for having an extra tool were the special case of GNAT proving, that ``run/vfs_cbe_init`` must be executed _before_ ``run/vfs_cbe`` and that the tests consume some time and that we maybe want to execute them only once a week.

m-stein, 2021-12-20:
I think, the first two of these problems can also be solved with run scripts.

m-stein, 2021-12-20:
Is ``Event`` service support desired for the black hole server? In order to overcome an unresolved session request in the "autopilot" mode of ``run/vfs_cbe`` originating from the drivers subsystem, I added this feature to the black hole server. Afterwards, I found that I can also just remove the whole drivers subsystem without breaking anything.

nfeske, 2021-12-21:
> Is Event service support desired for the black hole server?

That would be a very welcome addition. In fact, I think we should consider all prominent server interfaces to be covered by the black-hole component in the longer term.

nfeske, 2022-02-11:
Merged to staging.