genodelabs/genode #4357: foc: l4_thread_stats_time syscall does not return in core (cpu_balancer qemu x86_32/x86_64)

alex-ab, 2021-12-20:


chelmuth, 2021-12-21:
Is there any other thread running in space 11f?

A quick look into the sources reveals: The mysterious DRQ (I never heard of before) may correlate with cross-CPU IPC (or potentiallly other cross-CPU operations).

chelmuth, 2021-12-21:
I'd bet 
https://github.com/kernkonzept/fiasco/blob/4367c50b1df878604b9833d4b41ae311d6e12e03/src/kern/thread_object.cpp#L678-L684 is the source of the *deferred request (DRQ)* and triggers a corner case of remote CPU stats retrieval.

alex-ab, 2022-01-05:
I added one commit to disable the cpu_balancer test and another one which should prevent users of the trace framework to hang up the roottask on Genode/FOC due to the bug.

nfeske, 2022-01-05:
Thanks @alex-ab. I merged both commits along with the style fixup https://github.com/genodelabs/genode/commit/f4f7c1a5573462eeb753ab0ac6aa8b606eb48321.

commit d4c58b4e (2022-01-05): cpu_balancer.run: disable testing on foc

Due to some kernel issue the test will not finish, see #4357

Fixes #4357
commit 4ac553e5 (2022-01-19): cpu_balancer.run: disable testing on foc

Due to some kernel issue the test will not finish, see #4357

Fixes #4357
closed 2022-01-19 by chelmuth