genodelabs/genode #4359: Deprecate platform API specific to x86, imx53, rpi

skalk, 2021-12-22:
To cleanup the API, and to enable the usage of the newer - by now ARM only - platform driver on x86 or riscv, we shall deprecate the former platform APIs specific to x86, imx53, rpi by moving them to "legacy" subdirectory. The driver incarnations shall be renamed to "legacy_*_platform_drv" to be able to use them side-by-side with the new incarnation while drivers are updated gradually.

tomga, 2021-12-26:
@chelmuth, @nfeske, @skalk, may I ask to avoid merging rpi specific parts to staging at least before considering merge of my changes from issue #3415? I've finished cleanup of the code and was in process of describing [current state](/genodelabs/genode/issues/3415#issuecomment-1001232845) when I found out about this issue. This commit will produce conflicts with my changes. I also believe that after my changes old legacy rpi platform will not be needed anymore and will be easy to remove.

skalk, 2022-01-03:
@tomga I fully understand your worries about potential merge conflicts and additional burden on your side. I've tried to review your commit series as a whole but there are still open questions, or changes I would like to investigate with more care. Though, I'm afraid we cannot withhold commits - critical for further development in our daily work - to keep a merge window open.
Nonetheless, I like to offer my assistance to constantly and consequently bring your changes to staging step by step, begining from today. I'll have however ask you back for testing. I hope this is understandable and tolerable for you.

tomga, 2022-01-03:
@skalk, thanks. My comments below.

> I fully understand your worries about potential merge conflicts and additional burden on your side.

Actually I looked at your patch briefly and I was sure that there will be merge conflicts. Anyway I rebased my branch on top of your commits today and after some iteration of fixes everything seems to work. I'll write more details in #3415.

> I've tried to review your commit series as a whole but there are still open questions, or changes I would like to investigate with more care.

I'm aware of some shortcuts I made - they definitely need some care. I believe/hope that they are not critical and can be addressed later. I really did not want to keep all that code in my branch any longer (and chase your changes :-) ).
 
> I like to offer my assistance to constantly and consequently bring your changes to staging step by step, begining from today.

I really appreciate this and will try to make this process as easy as possible. For both of us.

> I'll have however ask you back for testing. I hope this is understandable and tolerable for you.

Yes. Of course. I have no problem with that.

skalk, 2022-01-04:
@tomga thank you for taking the time and for your understanding.

skalk, 2022-01-10:
@nfeske @chelmuth I've rebased the work again to match the changes of the staging branch. As this deprecation touches several configuration files that are also essential during driver upgrade work, it would be fine if you could review resp. merge the commit 6a0b6db when time permits it.

nfeske, 2022-01-10:
Thank you @skalk for the notice. I merged this very welcome change to staging now. It is nice to see some outdated spec/ directories vanish with it.

nfeske, 2022-01-11:
I adjusted the genode-allwinner repository in commit https://github.com/nfeske/genode-allwinner/commit/ac4c6e44307afae5c335fd3cbb06f9e549d48e1b.

commit bdb9df1d (2022-01-10): os: deprecate platform API for x86, rpi, imx53

* Move platform APIs to "legacy/" subdirectory
* Rename old pc, imx53, and rpi platform_drv
  to "legacy_*_platform_drv"

Fix #4359
closed 2022-01-19 by chelmuth