genodelabs/genode #4360: test-pthread timeouts on rpi3 if run after other tests with depot_autopilot

tomga, 2022-01-02:
During testing my [rpi dev branch](/tomga/genode/commits/rpi_master_20211217) I noticed that `test-pthread` fails with timeout after 90 seconds on rpi3. I have verified that this behavior exists also without my changes so I'm creating a new issue for that.

I did some additional tests and:
 * this timeout does not happen if `test-pthread` is as a single test using `depot_autopilot` - in this case it finishes in about 32s on rpi3
 * this does not happen on rpi1 (`arm_v6`) - test finishes in about 34.5 s no matter if run during whole testsuite run or with just a single test
 * I was curious if there is some single test that causes loss of performance of `test-pthread` test so I run it with different sets of tests that executed before and I found out that there wasn't one special test that caused timeout but each test run before `test-pthread` caused some performance penalty. Each added from 1 to few seconds to time required to finish `test-pthread`.
 * I did not check yet in detail what this test does but the period that gets longer is in place of dots in following log (`request shutdown` is never printed in case of timeout):
```
[init -> depot_autopilot] 12.729 [init -> test-pthread] waiter(main thread): started
[init -> depot_autopilot] 12.737 [init -> test-pthread] waiter: started
...
[init -> depot_autopilot] 29.065 [init -> test-pthread] waiter(main thread): request shutdown
[init -> depot_autopilot] 29.074 [init -> test-pthread] signaller: shutting down
```
 * it seems that only this one test and only this specific part of it has this problem of execution time becoming longer if executed after some other tests. Other tests and other parts of this test do not exhibit this behavior.

Can you check if that happens also on other boards with `arm_v7a` (I can observe this on rpi2 too) and `arm_v8a`? I'd like to know if it is specific to Raspberry Pi somehow or a generic problem.

Do you have any ideas what can cause such specific behavior?

One more thing. All tests that cause a timeout from GNU C++ testsuite mentioned in #3925 are somehow connected to *pthreads*. Is it possible that those this issue is related those C++ testsuite failures?

chelmuth, 2022-01-05:
I wonder if your observation that other tests *added from 1 to few seconds to time required to finish test-pthread* can be linked to the fact that *test-pthread* is started *later*. Could it be a time-drift issue? You may check by delaying the start of the test in *depot_autopilot* or the pthread test itself by some dozens of seconds.

tomga, 2022-01-06:
I've made some tests after adding sleep in `depot_autopilot` before running a single `test-pthread` test. Results are a little surprising. This delay increases time needed to finish pthread test but at much lower scale. Thirty minutes delay caused the test to run about 47 secons so about 15 seconds longer then when started without delay. But if it is run with default list of tests `test-pthread` starts after about 5 minutes after `depot_autopilot` starts and 90 seconds (timout value) is not enough to finish the test successfully.

So there is *something* that is *used/exhausted* slowly when `depot_autopilot` is sleeping and much quicker when it executes tests and that *someting* causes slowdown of specific part of `test-pthread` functionality.

I also made an attempt to wrap `depot_autopilot` into a `sequence` with an idea to run it in a loop to see if this *something* is inside the `depot_autopilot` or outside of it but I failed to properly change the routes in the run script to achieve this.

It would be useful to know if similar behavior can be observed on other `arm_v7a` or `arm_v8a` boards. I think that if it is reproducible then probably the problem may be too complex for me to tackle it. On the other hand if it is specific for Raspberry Pi then I'd try to dig into it more as others have probably more important things right now.

jschlatow, 2022-07-14:
I somehow overlooked this issue. Today, I executed `test-pthread` as a single test. On a Zynq board, the test finished only after 93s (I increased the test timeout). On zynq_qemu, the test finishes after 35s. 

The following lines when executed on real hardware arouse suspicion:
```
[init -> depot_autopilot] 1.701 [init -> test-pthread] main thread: start PTHREAD_MUTEX_NORMAL stress test
[init -> depot_autopilot] 5.356 [init -> test-pthread] thread 0x402fecb0: 800 rounds done
[init -> depot_autopilot] 5.360 [init -> test-pthread] thread 0x402fecc8: 800 rounds done
...
[init -> depot_autopilot] 5.454 [init -> test-pthread] main thread: start PTHREAD_MUTEX_ERRORCHECK stress test
[init -> depot_autopilot] 8.786 [init -> test-pthread] thread 0x402fecb0: 800 rounds done
[init -> depot_autopilot] 8.791 [init -> test-pthread] thread 0x402fecc8: 800 rounds done
...
[init -> depot_autopilot] 8.885 [init -> test-pthread] main thread: start PTHREAD_MUTEX_RECURSIVE stress test
[init -> depot_autopilot] 12.528 [init -> test-pthread] thread 0x402fecb0: 800 rounds done
[init -> depot_autopilot] 12.533 [init -> test-pthread] thread 0x402fecc8: 800 rounds done
...
[init -> depot_autopilot] 14.382 [init -> test-pthread] waiter: started
[init -> depot_autopilot] 87.284 [init -> test-pthread] waiter(main thread): request shutdown
[init -> depot_autopilot] 87.288 [init -> test-pthread] signaller: shutting down
[init -> depot_autopilot] 87.352 [init -> test-pthread] signaller: finished after 3576 state changes
[init -> depot_autopilot] 87.357 [init -> test-pthread] waiter(main thread): finished (pings=1004, pongs=996, iterations=2001)
[init -> depot_autopilot] 87.430 [init -> test-pthread] waiter: finished (pings=1047, pongs=946, iterations=1994)
[init -> depot_autopilot] 87.435 [init -> test-pthread] waiter: finished (pings=1000, pongs=992, iterations=1994)
...
[init -> depot_autopilot] 87.482 [init -> test-pthread] ad)waiter: started
[init -> depot_autopilot] 93.190 [init -> test-pthread] signaller: finished
...
```

Notice the long delay when setting up the mutex stress tests and the long execution time of the signalling phase.

tomga, 2022-07-18:
Thank you @jschlatow for taking time to verify this on non-rpi platform and confirming that it is not specific to rpi. Do you have any ideas about what the real problem may be?

jschlatow, 2022-07-18:
@tomga Not really, no. So far I checked whether there is an inaccuracy in `usleep()`, which is called by the signaller and waiter threads. I also used the `trace_subject_reporter` to have a look at the thread execution times. It seems as if the idle thread consumes more of the CPU time if another test was executed before. I don't have a clue at the moment, however, why the signaller and waiter threads make slower progress.

jschlatow, 2022-07-19:
I must correct my previous statement that `usleep()` is not the culprit. I only tested `usleep()` in a single thread before in which case it is not affected. However, if I run two threads that both perform `usleep()` concurrently, it **is** affected.

Note that `test-pthread` uses very short sleep times of 1ms and 3ms. If I execute a loop with 2000 iterations and `usleep(1000)`, it takes approx. 10 seconds to finish irrespective of whether another test has been executed before.

Yet, I came up with the following test case that exposes the problem:
```
static void *thread_sleep_func(void *arg)
{
	unsigned id = *(unsigned*)arg;
	Genode::log("sleeper thread ", id, " started");
	for (unsigned i=0; i < 2000; i++)
		usleep(1000);
	Genode::log("sleeper thread ", id, " finished");

	return nullptr;
}

static void test_sleep()
{

	unsigned arg1 = 1;
	unsigned arg2 = 2;

	pthread_t t1;
	pthread_create(&t1, 0, thread_sleep_func, &arg1);

	pthread_t t2;
	pthread_create(&t2, 0, thread_sleep_func, &arg2);

	pthread_join(t1, nullptr);
	pthread_join(t2, nullptr);
}
```

When I execute this as the sole test component with depot_autopilot, it takes approx. 40s to finish. When I execute `test-trace_logger` before, it takes 57s.

jschlatow, 2022-07-19:
Eureka! The problem lies within the timeout framework. I instrumented the code with `GENODE_LOG_TSC` and stumbled upon the fact that, after a few seconds execution time, there are more `handle_timeout` calls than `_schedule_timeout` calls. I thus added a warning in `handle_timeout` in case rate limiting is applied. Here is what happens:

```[init -> depot_autopilot] 
[init -> depot_autopilot] --- Run "test-pthread" (max 90 sec) ---
[init -> depot_autopilot] 
[init -> depot_autopilot] 0.214 [init -> test-pthread] --- pthread test ---
[init -> depot_autopilot] 0.216 [init -> test-pthread] main thread: running, my thread ID is 0x10fef420
[init -> depot_autopilot] 0.231 [init -> test-pthread] sleeper thread 1 started
[init -> depot_autopilot] 0.233 [init -> test-pthread] sleeper thread 2 started
[init -> depot_autopilot] 0.241 [init -> test-pthread] sleeper thread 1 iteration=0
[init -> depot_autopilot] 0.249 [init -> test-pthread] sleeper thread 2 iteration=0
[init -> timer] TSC _insert_into_timeouts_list: 350K (1000 calls, last 352)
[init -> timer] TSC _schedule_timeout: 5791K (1000 calls, last 1195K)
[init -> timer] TSC _set_time_source_timeout: 2010K (1000 calls, last 2066)
[init -> timer] TSC handle_timeout: 8544K (1000 calls, last 7228)
[init -> timer] TSC _insert_into_timeouts_list: 698K (2000 calls, last 309)
[init -> timer] TSC _schedule_timeout: 11575K (2000 calls, last 1188K)
[init -> depot_autopilot] 5.420 [init -> test-pthread] TSC _insert_into_timeouts_list: 366K (1000 calls, last 406)
[init -> depot_autopilot] 5.431 [init -> test-pthread] sleeper thread 2 iteration=500
[init -> depot_autopilot] 5.434 [init -> test-pthread] sleeper thread 1 iteration=500
[init -> depot_autopilot] 5.442 [init -> test-pthread] TSC _schedule_timeout: 42065K (1000 calls, last 54178)
[init -> depot_autopilot] 5.458 [init -> test-pthread] TSC _set_time_source_timeout: 19863K (1000 calls, last 20789)
[init -> depot_autopilot] 5.463 [init -> test-pthread] TSC handle_timeout: 27336K (1000 calls, last 26125)
[init -> depot_autopilot] 6.947 [init -> test-pthread] Warning: current time: 6688237 rate_limit_deadline: 6688238
[init -> depot_autopilot] 6.968 [init -> test-pthread] Warning: current time: 6708998 rate_limit_deadline: 6708999
[init -> depot_autopilot] 6.989 [init -> test-pthread] Warning: current time: 6729758 rate_limit_deadline: 6729759
[init -> depot_autopilot] 7.026 [init -> test-pthread] Warning: current time: 6766851 rate_limit_deadline: 6766852
...
```

The first >1000 iterations of the sleeper threads are running without any problems. Yet, at some point, we often run into the rate limiting case. This ultimately leads to a lot of `trigger_once` calls to the Timer component that slow down the `usleep()`. What puzzled me is that the rate limit deadline is **always** exactly 1us ahead of the current time if we land into the rate limiting case. (Note, the timer session has a rate limit period of 1us.) I suspected the current time not being updated correctly. When I disable the interpolation feature, the test runs smoothly and repeatedly finishes after ~10s.

I haven't looked into the interpolation code since I don't have an intuition what it is supposed to do. @m-stein Do you have an idea what might be wrong here?

m-stein, 2022-07-20:
@jschlatow I fear I'm too much out of context to help. It's a long time since I looked into time interpolation and rate limiting and would have to investigate myself. However, I'd like too stay with my other work this week. I could give you a general overview of interpolation or explain concrete places in the implementation though, if this helps.

jschlatow, 2022-07-21:
@m-stein no worries. I'll have a look at #2400 and the corresponding commit messages.

jschlatow, 2022-07-21:
I'm not sure yet whether/how this is related to the rate limiting issue but for some reason, the interpolated time is often ahead of the real time by approx 4-5ms. I'm on base-hw, where `Kernel::time()` is used as timestamp for interpolation. If I got it correctly, `Kernel::time()` is also used by the Timer for `elapsed_us()`, hence I'm wondering how that much of a difference can occur when the same time base is used. I will investigate further if time permits but if either of you has any explanation/suspicion, let me know.

m-stein, 2022-07-21:
As far as I remember, the timer uses the knowledge about the frequency of ``Kernel::time(); ``. The application-side timeout framework however doesn't make assumptions about the frequency but always tries to figure it out via interpolation.

m-stein, 2022-07-21:
Technically, we could use the fact that base-hw timestamp frequency is fixed in applications as well. The argument not to do this was, I think, that it would prevent full control of parents over the timing of their children.

chelmuth, 2022-08-09:
Does the effect vanish entirely if you apply the following patch?

```diff
+++ b/repos/libports/recipes/pkg/test-pthread/runtime
@@ -33,7 +33,9 @@
 			<resource name="RAM" quantum="64M"/>
 			<config>
 				<vfs> <dir name="dev"> <log/> </dir> </vfs>
-				<libc stdout="/dev/log"/>
+				<libc stdout="/dev/log">
+					<pthread placement="single-cpu"/>
+				</libc>
 			</config>
 		</start>
 	</config>
```

jschlatow, 2022-08-09:
@chelmuth yes, I forgot to mention that I've already tried this and that it circumvents the problem.

cproc, 2022-10-20:
- all CPUs idle except for the occasional time logging (the idle thread calls `nop` instead of `wfi` to let the cycle counter (trace timestamp) continue counting):

```
[init -> time_logger] cpu 0: timestamp():       16564273043 (diff to idle cpu 3: 35715)
[init -> time_logger] cpu 0: timestamp() in us: 20299354 us (diff to idle cpu 3: 44 us)
[init -> time_logger] cpu 0: curr_time():       20100554 us (diff to idle cpu 3: 0 us)
[init -> time_logger] cpu 0: elapsed_us():      20199128 us (diff to idle cpu 3: 11 us)
[init -> time_logger] cpu 0: Kernel::time():    37845120 us (diff to idle cpu 3: -52 us)
[init -> time_logger] 
[init -> time_logger] cpu 1: timestamp():       16564242340 (diff to idle cpu 3: 5012)
[init -> time_logger] cpu 1: timestamp() in us: 20299316 us (diff to idle cpu 3: 6 us)
[init -> time_logger] cpu 1: curr_time():       20100554 us (diff to idle cpu 3: 0 us)
[init -> time_logger] cpu 1: elapsed_us():      20199094 us (diff to idle cpu 3: -23 us)
[init -> time_logger] cpu 1: Kernel::time():    37845158 us (diff to idle cpu 3: -14 us)
[init -> time_logger] 
[init -> time_logger] cpu 2: timestamp():       16564238900 (diff to idle cpu 3: 1572)
[init -> time_logger] cpu 2: timestamp() in us: 20299312 us (diff to idle cpu 3: 2 us)
[init -> time_logger] cpu 2: curr_time():       20100554 us (diff to idle cpu 3: 0 us)
[init -> time_logger] cpu 2: elapsed_us():      20199106 us (diff to idle cpu 3: -11 us)
[init -> time_logger] cpu 2: Kernel::time():    37845163 us (diff to idle cpu 3: -9 us)
[init -> time_logger] 
[init -> time_logger] cpu 3: timestamp():       16564237328 (diff to idle cpu 3: 0)
[init -> time_logger] cpu 3: timestamp() in us: 20299310 us (diff to idle cpu 3: 0 us)
[init -> time_logger] cpu 3: curr_time():       20100554 us (diff to idle cpu 3: 0 us)
[init -> time_logger] cpu 3: elapsed_us():      20199117 us (diff to idle cpu 3: 0 us)
[init -> time_logger] cpu 3: Kernel::time():    37845172 us (diff to idle cpu 3: 0 us)
```

The time difference between the CPUs was small and stayed that way.

cproc, 2022-10-20:
- an additional "burner" component executed a busy loop on CPU 1:

```
top:

[init -> top] cpu=0.0   0     0  96.73%   0.00% thread='idle'                    label='kernel'
[init -> top] cpu=0.0   0     0   3.00%   0.00% thread='entrypoint'              label='core'
[init -> top] cpu=0.0   3     0   0.18%   0.00% thread='ep'                      label='init -> time_logger'
[init -> top] cpu=0.0   1     0   0.06%   0.00% thread='ep'                      label='init -> top'
[init -> top] cpu=0.0   3     0   0.00%   0.00% thread='timer'                   label='init -> timer'
[init -> top] cpu=0.0   3     0   0.00%   0.00% thread='ep'                      label='init -> timer'
[init -> top] cpu=1.0   1     0  99.99%   0.00% thread='ep'                      label='init -> time_burner'
[init -> top] cpu=1.0   3     0   0.00%   0.00% thread='time_logger'             label='init -> time_logger'
[init -> top] cpu=1.0   1     0   0.00%   0.00% thread='time_burner'             label='init -> time_burner'
[init -> top] cpu=1.0   0     0   0.00%   0.00% thread='idle'                    label='kernel'
[init -> top] cpu=2.0   0     0  99.99%   0.00% thread='idle'                    label='kernel'
[init -> top] cpu=2.0   3     0   0.00%   0.00% thread='time_logger'             label='init -> time_logger'
[init -> top] cpu=3.0   0     0  99.99%   0.00% thread='idle'                    label='kernel'
[init -> top] cpu=3.0   3     0   0.00%   0.00% thread='time_logger'             label='init -> time_logger'

time values after 10 seconds:

[init -> time_logger] cpu 0: timestamp():       8409238286 (diff to idle cpu 3: -11410)
[init -> time_logger] cpu 0: timestamp() in us: 10305439 us (diff to idle cpu 3: -14 us)
[init -> time_logger] cpu 0: curr_time():       10198036 us (diff to idle cpu 3: 30 us)
[init -> time_logger] cpu 0: elapsed_us():      10198050 us (diff to idle cpu 3: 25 us)
[init -> time_logger] cpu 0: Kernel::time():    21930135 us (diff to idle cpu 3: -85 us)
[init -> time_logger] 
[init -> time_logger] cpu 1: timestamp():       8418825020 (diff to idle cpu 3: 9575324)
[init -> time_logger] cpu 1: timestamp() in us: 10317187 us (diff to idle cpu 3: 11734 us)
[init -> time_logger] cpu 1: curr_time():       10209733 us (diff to idle cpu 3: 11727 us)
[init -> time_logger] cpu 1: elapsed_us():      10229739 us (diff to idle cpu 3: 31714 us)
[init -> time_logger] cpu 1: Kernel::time():    21981893 us (diff to idle cpu 3: 51673 us)
[init -> time_logger] 
[init -> time_logger] cpu 2: timestamp():       8409253316 (diff to idle cpu 3: 3620)
[init -> time_logger] cpu 2: timestamp() in us: 10305457 us (diff to idle cpu 3: 4 us)
[init -> time_logger] cpu 2: curr_time():       10197974 us (diff to idle cpu 3: -32 us)
[init -> time_logger] cpu 2: elapsed_us():      10197995 us (diff to idle cpu 3: -30 us)
[init -> time_logger] cpu 2: Kernel::time():    21930193 us (diff to idle cpu 3: -27 us)
[init -> time_logger] 
[init -> time_logger] cpu 3: timestamp():       8409249696 (diff to idle cpu 3: 0)
[init -> time_logger] cpu 3: timestamp() in us: 10305453 us (diff to idle cpu 3: 0 us)
[init -> time_logger] cpu 3: curr_time():       10198006 us (diff to idle cpu 3: 0 us)
[init -> time_logger] cpu 3: elapsed_us():      10198025 us (diff to idle cpu 3: 0 us)
[init -> time_logger] cpu 3: Kernel::time():    21930220 us (diff to idle cpu 3: 0 us)

time values after 10 minutes:

[init -> time_logger] cpu 0: timestamp():       493448026466 (diff to idle cpu 3: -24207)
[init -> time_logger] cpu 0: timestamp() in us: 604715718 us (diff to idle cpu 3: -30 us)
[init -> time_logger] cpu 0: curr_time():       604607258 us (diff to idle cpu 3: 29 us)
[init -> time_logger] cpu 0: elapsed_us():      604607272 us (diff to idle cpu 3: 25 us)
[init -> time_logger] cpu 0: Kernel::time():    616339357 us (diff to idle cpu 3: 620 us)
[init -> time_logger] 
[init -> time_logger] cpu 1: timestamp():       493464176448 (diff to idle cpu 3: 16125775)
[init -> time_logger] cpu 1: timestamp() in us: 604735510 us (diff to idle cpu 3: 19762 us)
[init -> time_logger] cpu 1: curr_time():       604626984 us (diff to idle cpu 3: 19755 us)
[init -> time_logger] cpu 1: elapsed_us():      604646989 us (diff to idle cpu 3: 39742 us)
[init -> time_logger] cpu 1: Kernel::time():    616399938 us (diff to idle cpu 3: 61201 us)
[init -> time_logger] 
[init -> time_logger] cpu 2: timestamp():       493448084813 (diff to idle cpu 3: 34140)
[init -> time_logger] cpu 2: timestamp() in us: 604715790 us (diff to idle cpu 3: 42 us)
[init -> time_logger] cpu 2: curr_time():       604607198 us (diff to idle cpu 3: -31 us)
[init -> time_logger] cpu 2: elapsed_us():      604607218 us (diff to idle cpu 3: -29 us)
[init -> time_logger] cpu 2: Kernel::time():    616338372 us (diff to idle cpu 3: -365 us)
[init -> time_logger] 
[init -> time_logger] cpu 3: timestamp():       493448050673 (diff to idle cpu 3: 0)
[init -> time_logger] cpu 3: timestamp() in us: 604715748 us (diff to idle cpu 3: 0 us)
[init -> time_logger] cpu 3: curr_time():       604607229 us (diff to idle cpu 3: 0 us)
[init -> time_logger] cpu 3: elapsed_us():      604607247 us (diff to idle cpu 3: 0 us)
[init -> time_logger] cpu 3: Kernel::time():    616338737 us (diff to idle cpu 3: 0 us)
```

There was a deviation between CPU 1 and the idle CPUs (the idle CPUs apparently delayed) which looks like a multiple of 10 ms and it went up another ~10 ms at some point between these two samples (not gradually) and then stayed at these values until I stopped the test. Maybe it is somehow related to scheduling although I configured a lower priority for the burner component, so not sure if this result is to be expected.

cproc, 2022-10-20:
- the burner component calls `Blockade::block()` in a loop on CPU 1 and `Blockade::wakeup()` in a loop on CPU 2:

```
top:

[init -> top] cpu=0.0   0     0  96.70%   0.00% thread='idle'                    label='kernel'
[init -> top] cpu=0.0   0     0   3.20%   0.00% thread='entrypoint'              label='core'
[init -> top] cpu=0.0   1     0   0.07%   0.00% thread='top'                     label='init -> top'
[init -> top] cpu=0.0   3     0   0.01%   0.00% thread='ep'                      label='init -> time_logger'
[init -> top] cpu=0.0   3     0   0.00%   0.00% thread='timer'                   label='init -> timer'
[init -> top] cpu=0.0   3     0   0.00%   0.00% thread='ep'                      label='init -> timer'
[init -> top] cpu=1.0   1     0  99.97%   0.00% thread='ep'                      label='init -> time_burner'
[init -> top] cpu=1.0   0     0   0.02%   0.00% thread='idle'                    label='kernel'
[init -> top] cpu=1.0   1     0   0.00%   0.00% thread='time_burner'             label='init -> time_burner'
[init -> top] cpu=1.0   3     0   0.00%   0.00% thread='time_logger'             label='init -> time_logger'
[init -> top] cpu=2.0   1     0  99.99%   0.00% thread='burn_helper'             label='init -> time_burner'
[init -> top] cpu=2.0   3     0   0.00%   0.00% thread='time_logger'             label='init -> time_logger'
[init -> top] cpu=2.0   0     0   0.00%   0.00% thread='idle'                    label='kernel'
[init -> top] cpu=3.0   0     0 100.00%   0.00% thread='idle'                    label='kernel'
[init -> top] cpu=3.0   3     0   0.00%   0.00% thread='time_logger'             label='init -> time_logger'

time values after 10 seconds:

[init -> time_logger] cpu 0: timestamp():       8409159932 (diff to idle cpu 3: -14695)
[init -> time_logger] cpu 0: timestamp() in us: 10305343 us (diff to idle cpu 3: -18 us)
[init -> time_logger] cpu 0: curr_time():       10197993 us (diff to idle cpu 3: 34 us)
[init -> time_logger] cpu 0: elapsed_us():      10198011 us (diff to idle cpu 3: 30 us)
[init -> time_logger] cpu 0: Kernel::time():    21834199 us (diff to idle cpu 3: -52 us)
[init -> time_logger] 
[init -> time_logger] cpu 1: timestamp():       8408924371 (diff to idle cpu 3: -250256)
[init -> time_logger] cpu 1: timestamp() in us: 10305054 us (diff to idle cpu 3: -307 us)
[init -> time_logger] cpu 1: curr_time():       10197924 us (diff to idle cpu 3: -35 us)
[init -> time_logger] cpu 1: elapsed_us():      10197947 us (diff to idle cpu 3: -34 us)
[init -> time_logger] cpu 1: Kernel::time():    21627780 us (diff to idle cpu 3: -206471 us)
[init -> time_logger] 
[init -> time_logger] cpu 2: timestamp():       8409161953 (diff to idle cpu 3: -12674)
[init -> time_logger] cpu 2: timestamp() in us: 10305345 us (diff to idle cpu 3: -16 us)
[init -> time_logger] cpu 2: curr_time():       10207942 us (diff to idle cpu 3: 9983 us)
[init -> time_logger] cpu 2: elapsed_us():      10217946 us (diff to idle cpu 3: 19965 us)
[init -> time_logger] cpu 2: Kernel::time():    21854227 us (diff to idle cpu 3: 19976 us)
[init -> time_logger] 
[init -> time_logger] cpu 3: timestamp():       8409174627 (diff to idle cpu 3: 0)
[init -> time_logger] cpu 3: timestamp() in us: 10305361 us (diff to idle cpu 3: 0 us)
[init -> time_logger] cpu 3: curr_time():       10197959 us (diff to idle cpu 3: 0 us)
[init -> time_logger] cpu 3: elapsed_us():      10197981 us (diff to idle cpu 3: 0 us)
[init -> time_logger] cpu 3: Kernel::time():    21834251 us (diff to idle cpu 3: 0 us)

time values after 10 minutes:

[init -> time_logger] cpu 0: timestamp():       491887931305 (diff to idle cpu 3: -230185)
[init -> time_logger] cpu 0: timestamp() in us: 602803837 us (diff to idle cpu 3: -282 us)
[init -> time_logger] cpu 0: curr_time():       602699645 us (diff to idle cpu 3: 34 us)
[init -> time_logger] cpu 0: elapsed_us():      602699662 us (diff to idle cpu 3: 29 us)
[init -> time_logger] cpu 0: Kernel::time():    614335849 us (diff to idle cpu 3: -295 us)
[init -> time_logger] 
[init -> time_logger] cpu 1: timestamp():       491873803576 (diff to idle cpu 3: -14357914)
[init -> time_logger] cpu 1: timestamp() in us: 602786523 us (diff to idle cpu 3: -17596 us)
[init -> time_logger] cpu 1: curr_time():       602699579 us (diff to idle cpu 3: -32 us)
[init -> time_logger] cpu 1: elapsed_us():      602699600 us (diff to idle cpu 3: -33 us)
[init -> time_logger] cpu 1: Kernel::time():    601909007 us (diff to idle cpu 3: -12427137 us)
[init -> time_logger] 
[init -> time_logger] cpu 2: timestamp():       491886037177 (diff to idle cpu 3: -2124313)
[init -> time_logger] cpu 2: timestamp() in us: 602801516 us (diff to idle cpu 3: -2603 us)
[init -> time_logger] cpu 2: curr_time():       602709661 us (diff to idle cpu 3: 10050 us)
[init -> time_logger] cpu 2: elapsed_us():      602719665 us (diff to idle cpu 3: 20032 us)
[init -> time_logger] cpu 2: Kernel::time():    614365305 us (diff to idle cpu 3: 29161 us)
[init -> time_logger] 
[init -> time_logger] cpu 3: timestamp():       491888161490 (diff to idle cpu 3: 0)
[init -> time_logger] cpu 3: timestamp() in us: 602804119 us (diff to idle cpu 3: 0 us)
[init -> time_logger] cpu 3: curr_time():       602699611 us (diff to idle cpu 3: 0 us)
[init -> time_logger] cpu 3: elapsed_us():      602699633 us (diff to idle cpu 3: 0 us)
[init -> time_logger] cpu 3: Kernel::time():    614336144 us (diff to idle cpu 3: 0 us)
```

There was a noticeable deviation of `Kernel::time()` on CPU 1. The cycle counters appeared to have slowed down a bit as well on CPUs 1 and 2.

jschlatow, 2022-10-20:
Have you checked whether your scenario may trigger a `wfe` instruction? Basically the same holds for the `wfe` as for the `wfi` instruction. At a first glance, `wfe` is only used by `Kernel::Lock::lock()`.

cproc, 2022-10-20:
- the burner component called `Blockade::block()` in a loop on CPU 1 and `Blockade::wakeup()` in a loop on CPU 2 as before and the `time_logger` component waited for 10 seconds before starting the measurements:

```
time values after 10 minutes:

[init -> time_logger] cpu 0: timestamp():       491179508113 (diff to idle cpu 3: -193375)
[init -> time_logger] cpu 0: timestamp() in us: 601935671 us (diff to idle cpu 3: -237 us)
[init -> time_logger] cpu 0: curr_time():       609038843 us (diff to idle cpu 3: 0 us)
[init -> time_logger] cpu 0: elapsed_us():      601831265 us (diff to idle cpu 3: 11 us)
[init -> time_logger] cpu 0: Kernel::time():    613466508 us (diff to idle cpu 3: -421 us)
[init -> time_logger] 
[init -> time_logger] cpu 1: timestamp():       491165647823 (diff to idle cpu 3: -14053665)
[init -> time_logger] cpu 1: timestamp() in us: 601918686 us (diff to idle cpu 3: -17222 us)
[init -> time_logger] cpu 1: curr_time():       598715481 us (diff to idle cpu 3: -10323362 us)
[init -> time_logger] cpu 1: elapsed_us():      601831242 us (diff to idle cpu 3: -12 us)
[init -> time_logger] cpu 1: Kernel::time():    598806984 us (diff to idle cpu 3: -14659945 us)
[init -> time_logger] 
[init -> time_logger] cpu 2: timestamp():       491177748400 (diff to idle cpu 3: -1953088)
[init -> time_logger] cpu 2: timestamp() in us: 601933515 us (diff to idle cpu 3: -2393 us)
[init -> time_logger] cpu 2: curr_time():       609038843 us (diff to idle cpu 3: 0 us)
[init -> time_logger] cpu 2: elapsed_us():      601831310 us (diff to idle cpu 3: 56 us)
[init -> time_logger] cpu 2: Kernel::time():    613476202 us (diff to idle cpu 3: 9273 us)
[init -> time_logger] 
[init -> time_logger] cpu 3: timestamp():       491179701488 (diff to idle cpu 3: 0)
[init -> time_logger] cpu 3: timestamp() in us: 601935908 us (diff to idle cpu 3: 0 us)
[init -> time_logger] cpu 3: curr_time():       609038843 us (diff to idle cpu 3: 0 us)
[init -> time_logger] cpu 3: elapsed_us():      601831254 us (diff to idle cpu 3: 0 us)
[init -> time_logger] cpu 3: Kernel::time():    613466929 us (diff to idle cpu 3: 0 us)
```

Now `curr_time()` deviated as well on CPU 1.

jschlatow, 2022-10-21:
That's good news...at least for the trace timestamp. I'm still wondering about the time drift of the kernel time though.

cproc, 2022-11-10:
I took a closer look at the `Kernel::time()` deviation and the cause appears to be a small gap between the end of a "duration" and the start of tne next one which is not accounted for in the total time calculation. Depending on how often `Timer::schedule_timeout()` gets called on each CPU, this gap can add up more or less. Commit f0357b8 fixes the deviation in the `time_logger` test on the PINE A64 board, but not sure if it fixes the original problem of this GitHub issue as well.

jschlatow, 2022-11-10:
And to answer your question about whether your fix solves this issue: We have to use the private timer on cortex a9 since, I believe, the generic timer is not implemented by the cortex a9. Therefore the original issue remains.

jschlatow, 2022-11-10:
AFAICT, there is a similar problem with the private timer implementation on cortex a9 (a small time gap remains unaccounted). I experimented a bit and found a minor improvement. The problem though is more conceptual as we cannot use the private timer as a monotone time source. There will always remain a small gap between reading the old timer value and setting a new one. 

I'm actually wondering whether we could use the global timer instead. Since the global timer is up-counting and provides a private comparator and interrupt per core, I believe we could use it as a kernel time source as well as for one shot timeouts similar to the generic timer.

tomga, 2022-11-10:
@cproc, @jschlatow, thank you for pursuing this.

> but not sure if it fixes the original problem of this GitHub issue as well.

I did some tests and in short: yes, it fixes original issue or Raspberry Pi 3

Below you have some more detailed results without and with latest fix proposal.

```
** before change

*** just test-pthread

[init -> depot_autopilot] 12.715 [init -> test-pthread] waiter(main thread): started
[init -> depot_autopilot] 12.723 [init -> test-pthread] waiter: started
[init -> depot_autopilot] 28.287 [init -> test-pthread] waiter: request shutdown
[init -> depot_autopilot] 28.294 [init -> test-pthread] signaller: shutting down

pc time: 15.59254,   rpi time: 15.564

[init -> depot_autopilot]  test-pthread                    ok        31.197  log

*** all tests

[init -> depot_autopilot] 12.792 [init -> test-pthread] waiter(main thread): started
[init -> depot_autopilot] 12.801 [init -> test-pthread] waiter: started
[init -> depot_autopilot] 59.592 [init -> test-pthread] waiter(main thread): request shutdown
[init -> depot_autopilot] 59.616 [init -> test-pthread] signaller: shutting down

pc time: 1:  46.791845,   rpi time: 46.791

[init -> depot_autopilot]  test-pthread                    ok        63.308  log

** with fix

*** just test-pthread

[init -> depot_autopilot] 12.713 [init -> test-pthread] waiter(main thread): started
[init -> depot_autopilot] 12.721 [init -> test-pthread] waiter: started
[init -> depot_autopilot] 28.285 [init -> test-pthread] waiter: request shutdown
[init -> depot_autopilot] 28.292 [init -> test-pthread] signaller: shutting down

pc time: 15.59329,   rpi time: 15.564

[init -> depot_autopilot]  test-pthread                    ok        31.146  log

*** all tests

[init -> depot_autopilot] 12.721 [init -> test-pthread] waiter(main thread): started
[init -> depot_autopilot] 12.729 [init -> test-pthread] waiter: started
[init -> depot_autopilot] 28.289 [init -> test-pthread] waiter: request shutdown
[init -> depot_autopilot] 28.297 [init -> test-pthread] signaller: shutting down

pc time: 15.589725,   rpi time: 15.56

[init -> depot_autopilot]  test-pthread                    ok        31.002  log
```

jschlatow, 2022-11-11:
@tomga Thanks for testing. I'm glad to hear that it fixes your issue with the rpi3. I have missed the fact that the rpi3 is an arm_v8a device when stating earlier that the fix does not apply to the original issue. Thanks for clarifying this.

chelmuth, 2022-11-14:
Is be1047eaf4ca19f021c313e9839a71a0f0bd9589 ready for a merge to staging? genode-riscv seems to adapt the same pattern. Is this ok?

cproc, 2022-11-21:
> Is [be1047e](https://github.com/genodelabs/genode/commit/be1047eaf4ca19f021c313e9839a71a0f0bd9589) ready for a merge to staging? genode-riscv seems to adapt the same pattern. Is this ok?

I think it's ready for now. Commit a5b2e77 applies the change to the riscv timer as well, but I haven't tested it yet.

jschlatow, 2023-01-19:
I have added an explanatory message to the aforementioned commit and integrated a fixup for pbxa9. The updated commit is 9a05d93.

@chelmuth from my perspective, the fix is ready to enter staging. Could you let me know whenever you merge it? I would like to update the zynq repo at the same time.

chelmuth, 2023-01-24:
@jschlatow I have no objections and will merge the commit after an "official" ack by @skalk :wink:

skalk, 2023-01-24:
@chelmuth ack!

chelmuth, 2023-01-25:
I merged ea9a15175a2b9dc13dddf8fe1d78d30af4e33a14  and genodelabs/genode-zynq@1891139a1700b041e9dc3fe28c0440443f6dacd2 to the staging branches. Can we close this issue now?

jschlatow, 2023-01-26:
@chelmuth Thanks. Yes, we can close this issue now.

closed 2023-01-26 by jschlatow