genodelabs/genode #4361: virtualbox6: support thread priorities

cproc, 2022-01-03:
Porting the thread priority support for Genode from virtualbox5 had a positive impact on audio playback.with Linux and Windows 10 VMs.

cproc, 2022-01-03:
Fixed by aae8b13

chelmuth, 2022-01-04:
I had an intense discussion with @nfeske about using your commit as an intermediate *sledgehammer* step on the way to a final solution without priorities as this is our definite goal. While debating about the execution model in VritualBox, the observed behavior, and potential causes, we identified only two priority levels that must be implemented in the end - one for the VMM and one for the VM/VCPU. For some reason, we missed to reduce the priority of the `Vm_connection` in the current implementation on staging, but you fixed this detail in your commit.

From this background, could you please test if it is sufficient to only reduce the `Vm_connection` priority in
https://github.com/genodelabs/genode/blob/aae8b13c5420150c73697c96525540583316d7ff/repos/ports/src/virtualbox6/sup_drv.h#L63-L64
to `Cpu_session::PRIORITY_LIMIT / 2` and drop the remaining patch?

cproc, 2022-01-04:
@chelmuth: yes, only reducing the 'Vm_connection' priority seems to have about the same positive effect.

For comparison, I ran three tests playing a 10 minute audio file with VLC in a Windows 10 VM and counted the number of detected underruns. First column is the number of priority levels.

```
 0: 1717 1977 1733
16:   83   58   80
 2:   69   76   61
```

Commit e2f5de5 implements the smaller fix.

chelmuth, 2022-01-05:
That sounds great, thanks. I merged 6bf81c4606d22b60419627299badcb2f8bef4c08 to staging.

chelmuth, 2022-01-05:
Now, there's the open question if https://github.com/genodelabs/genode/blob/master/repos/ports/src/virtualbox6/patches/devahci.patch and https://github.com/genodelabs/genode/blob/master/repos/ports/src/virtualbox6/patches/serial.patch are obsolete as VMM execution is effectively prioritized over VM/VCPU execution. The background is that these patches address cases were the guest code polls device-state changes but the VMM device model did not finish requests timely because the worker thread was delayed by the busy polling loop. Would you mind to check this theory by removing the patches?

cproc, 2022-01-07:
@chelmuth: I removed the two patches from my test setup and haven't noticed any problems related to that so far.

commit 6bf81c46 (2022-01-05): virtualbox6: reduce 'Vm_connection' priority

Fixes #4361
commit 5fed1641 (2022-01-19): virtualbox6: reduce 'Vm_connection' priority

Fixes #4361
closed 2022-01-19 by chelmuth