genodelabs/genode #4363: Move Raspberry Pi support into designated repository

skalk, 2022-01-04:
Analogue to other boards/SoC like imx and allwinner, the board support for Raspberry Pi variants shall be collected in one designated repositories.

skalk, 2022-01-10:
@nfeske @chelmuth as far as it matches your agenda, the removal from rpi1/3 from the genode main repository in commit dd61cb8 is ready to be merged. I've prepared a corresponding genode-rpi repository, which I would integrate into our testing infrastructure, as soon as you've merged this commit.

nfeske, 2022-01-10:
I merged the commit to staging as https://github.com/genodelabs/genode/commit/fe95e140944a68b4d1daa7130415cb25c36d79ef while mentioning the new location in the commit message.

chelmuth, 2022-01-19:
The issue is marked fixed and the commit entered master. Can we close this issue?

skalk, 2022-01-19:
Sure!

closed 2022-01-19 by skalk