genodelabs/genode #4366: tool/create_builddir: default config for linux can lead user to unexpected results

tomga, 2022-01-05:
Configuration created by default for `linux` architecture contains:
```
# create depot archives and update recipe versions automatically
#RUN_OPT += --depot-auto-update
```
and later:
```
RUN_OPT = --include boot_dir/linux \
          --include power_on/linux --include log/linux
```
Uncommenting `--depot-auto-update` does not make any effect because of the second part which resets value of RUN_OPT variable.

Commit 1bc1baadd13604df36567858e9d1b17c9a0b95ba changes second part to append options to RUN_OPT instead of resetting variable.

nfeske, 2022-01-06:
Thank you @tomga for bringing this issue to our attention. In fact, the `linux` argument of the `create_builddir` tool is a legacy artifact. I was actually surprised learning today that this option still exists. Linux can be targeted via the regular `x86_64`, `x86_32`, and `arm_v8a` build directories by now.

I would like to take this bug report as impulse to remove the `linux` argument from the `create_builddir` tool. Or do you see any use case for the argument that would be otherwise uncovered?

tomga, 2022-01-06:
I used this configuration for years and it generally worked with this one caveat so I finally mentioned it here. I didn't even know that `linux` can be targeted via architecture specific build directories (I didn't think about it really).

Therefore I have no objections to removal support for `linux` argument from `create_builddir` tool.

nfeske, 2022-01-06:
Thank you @tomga for the confirmation. Commit https://github.com/genodelabs/genode/commit/425a1426c0c646c332779e000f1ac59c99466515 removes the option from the `create_builddir` tool.

commit 425a1426 (2022-01-06): create_builddir: remove 'linux' option

Since unifying the build directories accross architectures, the special
case of the 'linux' platform is no longer needed. Linux can be targeted
from regular CPU-architecture-specific build directories via:

  KERNEL=linux BOARD=linux

Fixes #4366
closed 2022-01-19 by chelmuth