genodelabs/genode #4368: run tool: mechanism to obtain list of build artifacts

nfeske, 2022-01-07:
It would be nice if the run tool provided a way to track the build artifacts for invocations of the build system to be able to automate the creation of the list of boot modules for a given build. This way, the manually curated list of boot modules in run scripts could be replaced with a line like:

```
build_boot_image [build_artifacts]
```

This would simplify run scripts and make the combination of `import_from_depot` with custom built targets more convenient and robust.

nfeske, 2022-01-07:
This topic is related to a prior discussion in #4073.

tomga, 2022-01-07:
@nfeske, I did not realize that during our earlier discussion but I think it is not compatible with `--skip-build` option of `run` tool. So it won't be possible to use this option for those run scripts that use `build_artifacts` procedure? Am I right?

nfeske, 2022-01-07:
> Am I right?

That sounds logical. With no build, there are no build artifacts after all.

commit 275786a1 (2022-01-07): tool/run: way to obtain list of build artifacts

The new function 'build_artifacts' returns a list of artifacts created
by the Genode build system. The list can be supplied as argument to
the 'build_boot_image' function.

Note that the list covers only program targets and shared libraries.
Other artifacts created as side effects of custom rules are not covered.

Fixes #4368
closed 2022-01-19 by chelmuth