genodelabs/genode #4369: Modularization of Sculpt OS system images

nfeske, 2022-01-07:
Even though Sculpt is extremely flexible in terms of runtime customization, all Sculpt systems are currently based on one canonical system image, which was primarily designed for general-purpose computing. However, the architecture of Sculpt OS could be leveraged in other usage scenarios as well, in particular:

* Using Sculpt OS as testbed during development as an alternative to the use of run scripts for setups that require a mix of various device drivers.
* Special-purpose appliances with a deliberately narrow feature set such as storage-less systems that do not touch persistent storage at all.
* Turnkey solutions that do not feature any form of administrative user interface.

Even though such scenarios can already be accommodated by Sculpt, they usually call for manual patching and manual configuration, which stands in the way of being reproducible and maintainable.

nfeske, 2022-01-07:
Implemented in https://github.com/genodelabs/genode/commit/8ae74153591a7e65c3bbe0d240cd3e010aa9b561. Please refer to the commit message for more details.

Since the change is quite invasive, I'd be happy about people trying it out and confirm that I did not break anything. Suggestions in any directions are welcome.

One thing that I have not yet implemented is running `xmllint` on all ingredients. This would be a welcome addition.

chelmuth, 2022-01-11:
I love the new approach and tested it against my sculpt release workflow. Thereby, I identified some imperfections that may be addressed.

Up to the changes in this issue I could generate all depot archives required for sculpt distribution by simply running the following. The result also matched all requirements to build sculpt images according to the documentation at https://genode.org/documentation/articles/sculpt-21-10#Building_the_boot_image. 

```
./tool/depot/create UPDATE_VERSIONS=1 FORCE=1 -j64 genodelabs/bin/x86_64/base-nova genodelabs/pkg/x86_64/sculpt-pc genodelabs/pkg/x86_64/sculpt-installation
./tool/depot/publish_current -j4 genodelabs/bin/x86_64/base-nova genodelabs/pkg/x86_64/sculpt-pc genodelabs/pkg/x86_64/sculpt-installation
```

Unfortunately, *sculpt_distribution-pc* misses packages that were part of *sculpt-pc*, namely nvme_drv and intel_gpu_drv, and, thus, cannot replace sculpt-pc entirely. What's your opinion - should we add these two drivers to drivers_managed-pc? 

Also, the commit message in 8ae74153591a7e65c3bbe0d240cd3e010aa9b561 states

>  The former pkg/sculpt-installation and pkg/sculpt-installation-pc archives have been replaced by pkg/sculpt_distribution-pc, which references the generic pkg/sculpt_distribution archive.

which is not the case currently. So, *pkg/chroot* and *pkg/rom_filter* were also missing on `make -C build/x86_64 run/sculpt KERNEL=nova`.

With these tiny glitches worked-around, the new approach works like a charm.

nfeske, 2022-01-13:
@chelmuth thanks a lot for your feedback!

> Unfortunately, sculpt_distribution-pc misses packages that were part of sculpt-pc, namely nvme_drv and intel_gpu_drv, and, thus, cannot replace sculpt-pc entirely. What's your opinion - should we add these two drivers to drivers_managed-pc?

Yes, that is the right solution. I added the fixup c81048931714d46fc4d49e52163aecd535fb886f.

Sorry about the incomplete dependencies of pkg/sculpt_distribution-pc. To solve this issue, I would propose to add pkg/sculpt to pkg/sculpt_distribution/archives and replace pkg/sculpt by pkg/sculpt_distribution in the archives file of pkg/sculpt_distribution-pc. I would appreciate you trying this out.

chelmuth, 2022-01-13:
Thanks, the fixup and proposed changes to *pkg/sculpt_distribution/archives* and *pkg/sculpt_distribution* address my concerns entirely. It will work like a charm and look even more easy ;-)

```
./tool/depot/create UPDATE_VERSIONS=1 FORCE=1 -j64 genodelabs/pkg/x86_64/sculpt_distribution-pc
./tool/depot/publish_current -j4 genodelabs/pkg/x86_64/sculpt_distribution-pc
./tool/depot/publish genodelabs/index/<VERSION>
```

What's left is the cleanup of *default-pc.sculpt* which no longer need to declare `import: src/nvme_drv src/intel_gpu_drv` in addition to `pkg/drivers_managed-pc`. I can prepare a patch later by myself.

chelmuth, 2022-01-19:
Commit 2a9e3e1d86c0ae109090a196fcdfe241e84a2d85 adds the cleanup of *default-pc.sculpt*.

ssumpf, 2022-02-14:
@nfeske: Please consider a8bf355, cca80b5, and [ac332e3](https://github.com/ssumpf/genode/commit/ac332e3d450d6251b8914a3c41de5184faba40c9) for staging. The commits are the prerequisite for https://github.com/genodelabs/genode-world/issues/284 to work.

ssumpf, 2022-02-14:
There are fixes addressed above.

nfeske, 2022-02-15:
I added 14b93c5ff3f25933e2f27382a8477174bb7fdf76 instead of https://github.com/genodelabs/genode/commit/cca80b56e873249d671a3dc803a008998dc38abd, which results in an empty tar archive instead of no tar archive to avoid problems down the road when mounting depot.tar in ram fs. It also solves the problem more broadly, should this corner case ever appear in other run scripts.

nfeske, 2022-02-15:
I merged a slightly modified version of https://github.com/genodelabs/genode/commit/a8bf355856a664ba5966a9f909379f3156df09dc as 8edb7b28a00e23f9fb010e575948c1d7d1a99c50. The modification avoids returning an `Xml_node` as return value (which is a pointer, so better to avoid). @ssumpf are you fine with the new version under your name?

Could you give the boot2java sculpt another try with the current staging branch?

nfeske, 2022-02-28:
The feature is part of Genode 22.02.

closed 2022-02-28 by nfeske

chelmuth, 2022-03-08:
@nfeske Unfortunately, I missed to add a commit for your sketch in https://github.com/genodelabs/genode/issues/4369#issuecomment-1012102844. Commit cafb87d540a07692cbd49af59e1fe21fbc813573 catches up on this. Please note, I had to remove pkg/backdrop from sculpt/archives because of dependency detection. If you're okay with this, I'll merge the commit to staging.