genodelabs/genode #4372: vfs: support 'File_content' use with already opened file

cproc, 2022-01-11:
The ```File_content``` class is useful for accessing the XML content of the OSS VFS plugin's ```info``` file, but with the current implementation the info file gets opened repeatedly every few milliseconds when using audio with VirtualBox 6, which can be costly if the VFS uses plugins with possibly longer response times, for example the ```Fs``` plugin. One reason is that ```Dir_file_system::open()``` tries to determine if the given path is a directory by asking each VFS plugin for the directory status until one of them returns true, which is not the case for the info file. A simple solution for the OSS use case would be if the ```File_content``` class would also accept a reference to a ```Readonly_file```, which could be kept open and reused.

cproc, 2022-01-11:
Commit e354634 adds the alternative constructor.

nfeske, 2022-01-11:
The feature you are suggesting sounds useful. But I would rather avoid adding a second constructor to the `File_content` class but add a new function outside any class. Since the state - the file descriptor - is kept at the caller side, there is no need for an object.

```
template <typename FN>
void with_xml_file_content(Allocator &alloc, Readonly_file const &file, Size_to_read size_to_read, FN const &fn);
```

The functor `fn` would take an `Xml_node const &` as argument, like the `File_content::xml()` does.

The `File_content::xml()` method could then be changed to use this function.

@cproc Do you think this is feasible?

cproc, 2022-01-11:
@nfeske: i'm not sure what to do about the content buffer with your suggested solution. Currently, ```File_content``` fills an internal buffer and then the different methods work with this buffer. The new outside function would need to allocate and fill a content buffer on its own, right?

nfeske, 2022-01-12:
> The new outside function would need to allocate and fill a content buffer on its own, right?

Originally, I was thinking of a temporary allocation that is freed on return. However, seeing your use case, you would likely avoid dynamic allocations.

You are right. The `Allocator &` should better be a pointer to a local buffer. I think this tighter interface would be even better because it makes the function interface more rigid (there can't be any memory-allocation side effect of calling it). As a plus, the client can keep a static buffer around for repeated use without using a custom `Allocator` implementation.

Coding-style-wise, I would express this pointer as an object like this:

```
struct Byte_range_ptr
{
  char  * const start;
  size_t const num_bytes;

  Byte_range(char *start, size_t num_bytes)
  ...
};
```

The extra hoop of an additional object is good because it makes the caller think twice about the roles of the buffer size and the `size_to_read` parameter. The `const` members make the object non-copyable, which is a nice a precaution that encourages the caller to avoid keeping a copies of `Byte_range_ptr` objects (potentially dangling `start` pointers).

cproc, 2022-01-13:
Passing a pointer to an external buffer instead of an allocator sounds good to me. Now the other related question is how to best avoid the code duplication (and duplicated code execution) when reading the file content into the buffer. Currently, the ```File_content``` constructor reads the file content into a buffer. The new ```with_xml_file_content()``` function would read the file content into the given buffer as well. The code for reading the file content into a buffer could be moved into a separate function to avoid the code duplication, but if ```File_content::xml()``` would use the new ```with_xml_file_content()``` function, the file content would be read into the buffer for the second time.

nfeske, 2022-01-13:
> The code for reading the file content into a buffer could be moved into a separate function to avoid the code duplication, but if File_content::xml() would use the new with_xml_file_content() function, the file content would be read into the buffer for the second time.

How do you think about moving the file-read code from the `File_content` constructor to a new function `with_raw_file_content` that mirrors the interface of `with_xml_file_content`? This function would pass a `char const *ptr, size_t num_bytes` to the lambda (same pattern as `Xml_node::with_raw_content`). Note that in this case, the use of `Byte_range_ptr` would not add any benefit because the `ptr` is captured within the scope of the lambda function anyway.

With the `with_raw_file_content` function in place, the `File_content` constructor would become very simple.

I would not care too much about the remaining few similarities between `File_content::xml` and `with_xml_file_content` because this code does not implement any logic (except for the fallback to `<empty/>`, which should eventually go into the `Xml_node` constructor later on).

cproc, 2022-01-13:
Commit c328928 adds the new functions and commit 7f91258 adapts the OSS use case accordingly.

nfeske, 2022-01-14:
Thank you for the clean commit https://github.com/genodelabs/genode/commit/c328928176cacde05543dadbf41f5814fdd8eb4f. I request two changes though:

* Please revert the changes of the `File_content` methods and just call `with_raw_file_content` from the constructor. This avoids re-reading the file on each call of the accessor functions and keeps the patch small.
* The `Byte_range_ptr` struct should better go base/include/util/string.h, presumably somewhere at the upper part of the file. This way, we can start using it for everywhere throughout Genode where appropriate, not just in with the VFS.

cproc, 2022-01-14:
Updated commit cbe57b7 has the requested changes.

nfeske, 2022-01-17:
Thanks @cproc for implementing the changes. I merged the latest version to staging, supplemented with the minor fixup `978644369796bfb0c1e2a769eadd79637bb4ee02`. Your patience is much appreciated. :-)

commit bd0125ce (2022-01-17): vfs: add 'with_xml_file_content()' and 'with_raw_file_content()'

Fixes #4372
closed 2022-01-19 by chelmuth