genodelabs/genode #4380: intel/gpu: add export/import of buffer objects

ssumpf, 2022-01-13:
* add export/import of buffer objects from/to GPU session and GPU multiplexer
* create VFS plugin in order to wait for GPU session completion signal from non-EP threads
* libdrm: implement GEM contexts using multiple GPU sessions (from VFS plugin) and sharing buffer objects using import/export
* VritualBox6 initial 3D support

ssumpf, 2022-01-14:
@chelmuth: The above commit series reflects the current state of the Intel GPU driver, mesa, libdrm, and VBox6 on x86_64 with great performance improvements in the VBox6 case (~70% of native Genode 3D performance). Please have a look and consider for `staging`.

chelmuth, 2022-01-20:
I'm afraid fdd557d7f750256b5f6a5d4ef660695d93a749ca breaks https://github.com/skalk/genode-imx/blob/9b03c8bc59ea7c5af5ce93e717168a836ba11270/src/drivers/gpu/imx8mq/main.cc#L581 due to pure virtuals `export_buffer/import_buffer()`.

cnuke, 2022-01-20:
Commit e6fa257 in the *imx* repo address the problem.

ssumpf, 2022-01-20:
@chelmuth: c42b951 fixes session destruction in `intel_gpu_drv`, 49eae15 checks if '/dev/gpu' is present in VFS and prints an error message in case it is not.

chelmuth, 2022-01-20:
Thanks, I merged and tested both commits successfully.