genodelabs/genode #4381: virtualbox6 audio issues

cproc, 2022-01-13:
When playing back audio with VirtualBox 6, the OSS VFS plugin often runs out of audio data to submit to the audio driver, which causes noticeable dropouts. It turned out that the timeout handler of the HDA audio device model is often called too late. With the help of trace logs with timestamps I could identify several reasons so far:

- Timeout signals can occur too late (@chelmuth might know more details).
- Opening the 'info' file of the OSS VFS plugin repeatedly can take too long (issue #4372).
- A call of ```_fb.refresh()``` triggered by the VGA device model timeout handler can take too long.
- The ```write()``` libc monitor function can take too long when VirtualBox writes into the vdi file.
- The CPU (core) responsible for handling the VirtualBox timeouts can be running the vCPU too long. Apparently, on a Linux host the VMX preemption timer is programmed to leave the vCPU at the next expiring timeout. On Genode, the vCPU is paused either when the guest triggers an exit or when a different CPU is executing its EMT thread and decides to pause the vCPU, or when the VirtualBox watchdog thread ("Timer") becomes active every 10ms and detects that a timeout has expired. The interval of the watchdog timer can be lowered in the vbox file (```<ExtraDataItem name="VBoxInternal/TM/TimerMillies" value="10"/>```), which seems to help a bit, but might also increase the CPU load. It might also be possible to let the watchdog thread (or a similar new thread) block until the next timer expires instead of a fixed interval.
- ```pthread_cond_signal()``` does a handshake with signalled threads, which can take too long, although I haven't seen this problem anymore since the lowering of the vCPU priority (issue #4361).

cproc, 2022-01-17:
Commit 950f450 adapts the watchdog timer interval dynamically before entering the vCPU to wake it up close to the next expiring 'virtual sync' timeout instead of the fixed 10ms interval.

cproc, 2022-01-17:
In case the audio latency is perceived too high with some VMs or for some use cases, it is possible to reduce the size of the input and output pre-buffers in VirtualBox, which can hold up to 200ms by default, in the vbox file:

```
<ExtraDataItem name="VBoxInternal/Devices/hda/0/LUN#0/Config/In/PreBufferSizeMs" value="200"/>
<ExtraDataItem name="VBoxInternal/Devices/hda/0/LUN#0/Config/Out/PreBufferSizeMs" value="200"/>
```

nfeske, 2022-03-15:
The results of this work already went to the master branch.

closed 2022-03-15 by nfeske