genodelabs/genode #4384: uplink_client_base: fix deref of unconstructed connection

m-stein, 2022-01-19:
Imagine receiving the signal for an available TX ack or an available RX packet at the Uplink connection but a later received signal for a link-state change (to link state "down") at the same connection is handled first and destructs the Uplink connection before the handling of the former signals. In this case, the methods ``Uplink_client_base::_conn_tx_handle_ack_avail`` and ``Uplink_client_base::_conn_rx_handle_packet_avail`` must be guarded against an unconstructed ``_conn`` member, but they aren't so far.

m-stein, 2022-01-19:
Fixed:
cbe822f94b uplink_client_base: no deref of invalid connection
e99a7546f3 uplink_client_base: remove unused code

jschlatow, 2022-01-20:
@m-stein, the NIC driver in the genode-zynq repo still makes use of the custom handlers. The reason for this is that I followed a zero-copy approach for the driver which uses the packet buffers as DMA buffers. For this, I needed a way to reset a DMA descriptor only when the corresponding packet was acked and used the custom handlers for that (#4277).

Yesterday, I also discovered another use case for custom handlers (#4387). In short, I was trying to emulate blocking read/write interface. The read/write is executed in 3rd-party code that runs mutliple application threads. I therefore needed a way to notify the threads that they are ready to submit or that packets can be received and added more custom handlers.

What's your opinion on this, do you see alternatives to custom handlers in these use cases? In the second case at least, I'd be okay for not using Uplink_client_base since this is quite different from the typical NIC driver use case anyway.

m-stein, 2022-01-20:
@jschlatow I'm sorry, I didn't consider the Zynq driver as I forgot that it moved out of world. Do you need all of the original custom handlers plus your ``_custom_conn_tx_ack_avail_handler``? Regarding the tap scenario, I also doubt whether it's worth using the generic Uplink_client_base if you'd have to customize almost everything in the end. In such a situation I'd create a local copy of the Uplink_client_base, modify it for my own use case. Only after having finished my project I'd try to get rid of the remaining redundancies between my adaption and the original code. What do you think?

jschlatow, 2022-01-20:
@m-stein, no worries. The zynq driver uses `_custom_conn_tx_ack_avail_handler` and `_custom_conn_rx_packet_avail_handler`.

Regarding the tap scenario, I'll actually investigate what I end up with when writing the vfs_tap plugin. Thus you can ignore #4387.

nfeske, 2022-01-20:
@jschlatow As a suitable starting point for implementing a VFS directory structure of pseudo files, let me recommend the terminal_vfs_plugin (https://github.com/genodelabs/genode/blob/master/repos/os/src/lib/vfs/terminal_file_system.h) in combination with the libc support in commit 7ac32ea60c703a73477c121ab8269dfad201f428 as reference.

jschlatow, 2022-01-21:
@nfeske Thanks a lot for the pointer.

m-stein, 2022-01-25:
@jschlatow No, I don't have special plans regarding this. It would be nice if you could just add a fixup that suits your use-case :)

jschlatow, 2022-01-26:
Commit 746997d adds the missing custom handlers.

@m-stein: Do you have anything to add?
@chelmuth: Can you add this commit to staging?

nfeske, 2022-01-26:
Merged to staging.

m-stein, 2022-01-27:
@jschlatow Looks good, I have nothing to add. Thanks ;)

m-stein, 2022-01-28:
Sorry, 746997d looks good to me! Thanks!

m-stein, 2022-01-28:
Haha, played myself, thanks to an outdated version of the page in my browser ^_^

commit 94121e7c (2022-01-19): uplink_client_base: no deref of invalid connection

Imagine receiving the signal for an available TX ack or an available RX packet
at the Uplink connection but a later received signal for a link-state change
(to link state "down") at the same connection is handled first and destructs
the Uplink connection before the handling of the former signals. In this case,
the methods 'Uplink_client_base::_conn_tx_handle_ack_avail' and
'Uplink_client_base::_conn_rx_handle_packet_avail' must be guarded against an
unconstructed '_conn' member, but they weren't so far.

Fixes #4384
closed 2022-02-15 by nfeske