genodelabs/genode #4387: uplink_client_base: complete customisation of signal handlers

jschlatow, 2022-01-19:
For porting a 3rd-party library that uses Linux tap devices, I want to emulate an interface with blocking read/write operations using an uplink client as a backend. This use case actually provides the motivation to add customisability for the two remaining signal handlers, i.e. ready_for_ack and ready_to_ack.

chelmuth, 2022-01-20:
Please check your addition in combination with #4384. The corresponding commits already entered staging and seem to remove code you're using in cadence_gem.

nfeske, 2022-01-20:
@jschlatow How do you think about a new vfs_tap plugin that would make a (configurable) Uplink or Nic session available to the VFS and thereby the libc? This way, we could emulate the regular tap interface (letting the libc map the relevant ioctls to pseudo-file operations as we do for OSS) and use existing applications as is. The blocking semantics are taken care of by the libc.

jschlatow, 2022-01-20:
@nfeske, I briefly thought into this direction as well but opted for the quick-and-not-so-clean solution since I felt there was a bit too much uncertainty regarding how long it would take to write a vfs_tap and whether it would work in the end. If you think that it's worth spending time on this, I'd be happy to make an effort.

jschlatow, 2022-01-20:
On a second thought, writing a vfs_tap plugin might make a good example for a simple and illustrative genodians story about how to write vfs plugins with ioctl emulation via pseudo files.

chelmuth, 2022-01-20:
I'd love to test such a plugin with the VirtualBox tap backend...

nfeske, 2022-01-20:
> If you think that it's worth spending time on this, I'd be happy to make an effort.

That would be a definite definitely! In fact, I remember that we repeatedly encountered situations where we would have loved to have a VFS tap plugin at hand, but sighed with somewhat sad facial expression because it wasn't.

jschlatow, 2022-01-27:
Closing this issue in favor of #4394.

closed 2022-01-27 by jschlatow