genodelabs/genode #4390: Remove implicit blocking semantics from packet stream

nfeske, 2022-01-24:
The following packet-stream operations block implicitly under certain conditions:

* `get_packet` blocks until a packet becomes available in the submit queue,
* `submit_packet` blocks if the submit queue is full until,
* `acknowledge_packet` blocks if the ack queue is full, and
* `get_acked_packet` blocks if the ack queue is empty

When the packet-stream mechanism was originally introduced (https://genode.org/documentation/release-notes/9.11#Packet-stream_interface), Genode components were predominately designed as programs using blocking I/O. This practice has been largely replaced later (https://genode.org/documentation/release-notes/16.05#The_great_API_renovation) by the current notion of modelling components as state machines that handle I/O in an asynchronous fashion. The blocking semantics of the packet-stream mechanism are a relic from the early days of Genode.

We should now take the opportunity for simplification and remove the implicit blocking from the packet stream. This way, we gain two things:

1. The implementation of the blocking operations relies on dedicated `Signal_receiver` objects embedded in the packet-stream classes. Those signal receivers are constructed using a deprecated interface, which obscures the fact that global side effects can occur without explicitly specifying an entrypoint as blocking mechanism. This, in turn, stands in the way of coupling signals to entrypoints at the API level, which would allow for optimizing Genode's signal-delivery mechanism.

2. In modern components, the I/O operations are not expected to block if implemented correctly. As the packet stream blocks implicitly, however, implementation bugs (like calling `get_packet` without checking for `ack_avail` before) may remain hidden.

Given this rationale, I'd like to replace the calls of `wait_for_signal` by an error message and raising an exception.

This may deliberately break components that still (unknowingly) rely on the blocking semantics, so that we can update those, if present. A possible short-term fix for those components would be to add an explicit wait-and-dispatch-one-io-signal-loop before calling a (potentially) blocking function. But ultimately, such code would call for redesign.

nfeske, 2022-01-25:
Commit https://github.com/genodelabs/genode/commit/49dc02f31aca63b2455f55f756e2bc60732dca52 implements the change. At this point, I'm keeping the original function signatures in tact but throw an error instead of blocking. Down the road, it would be best to only keep the `try_*` methods.

nfeske, 2022-01-27:
I have now adjusted almost all components affected by this change.

As one remaining issue, libusb must be updated. Right now, the smartcard test produces the following error. I suspect similar behavior from other libusb clients.

```
[init -> test-smartcard] Error: attempt to retrieve packet from empty acknowledgement queue
[init -> test-smartcard] Error: Uncaught exception of type 'Genode::Packet_stream_source<Genode::Packet_stream_policy<Usb::Packet_descriptor, 64u, 64u, char> >::Empty_ack_queue'
[init -> test-smartcard] Warning: abort called - thread: ep
```

@cproc Since you are working in libusb right now, could you do the needed adjustment as you go? I'd very much appreciate that because I'm not familiar with libusb and also don't want to run into conflicts with your current line of work.

cproc, 2022-01-28:
Commit acc500f adjusts libusb.

nfeske, 2022-01-31:
Thanks @cproc. I merged your commit to staging.

commit 49dc02f3 (2022-01-25): os: remove blocking semantics from packet stream

Fixes #4390