genodelabs/genode #4391: nova: kernel pagefault using AMD+SVM

alex-ab, 2022-01-25:
```
NOVA Microhypervisor v8-1d52bd0 (x86_64): Jun 30 2021 16:56:38 [gcc 10.3.0] [MBI2]
[ 0] TSC:2997699 kHz BUS:0 kHz (measured) DL
[ 0] CORE:0:0:0 17:60:1:0 [0] AMD Ryzen 5 4600H with Radeon Graphics
...
Genode sculpt-21.03-234-g1d52bd017
...
[init -> vbox] main     Executable: /virtualbox
[init -> vbox] Error: SystemPropertiesWrap : Cannot determine default Guest Additions ISO location. Most likely they are not available
[init -> vbox] i_saveSettings skipped
[init -> vbox] Using SVM virtualization extension.
[init -> vbox] Warning: unsupported ioctl (request=0x4004746a)
[init -> vbox] Warning: unsupported ioctl (request=0x802c7414)
[init -> vbox] Warning: GIM: Warning!!! Host TSC is unstable. The guest may behave unpredictably with a paravirtualized clock.
[init -> vbox] Warning: unsupported ioctl (request=0x4004746a)
[init -> vbox] fb resize : [0] 1024x768 -> 720x400 (host: 1024x768)
[init -> vbox] fb resize : [0] 720x400 -> 640x480 (host: 1024x768)
[init -> vbox] Warning: attempt to handle the same signal context twice (nested)
[init -> vbox] fb resize : [0] 640x480 -> 720x400 (host: 1024x768)

[ 0] Killed EC:0xffffffff8601e040 SC:0xffffffff8699f040 V:0xe CR0:0x102490 CR3:0x0 CR4:0x2 (#PF (kernel))
```

The reported setup was runinng Genode/NOVA nested virtualized on Linux+KVM (5.14.18-100.fc33.x86_64) with qemu-5.1.0-9.fc33.

```
qemu-system-x86_64 -display sdl -cpu core2duo -m 8536 -accel kvm -cpu host -netdev bridge,id=net0,br=virbr0 -device e1000,netdev=net0 -drive id=disk,file=bin/ext2.img,format=raw,if=none -device ahci,id=ahci -device ide-hd,drive=disk,bus=ahci.0 -boot d -serial mon:stdio -m 8536 -cdrom var/run/vbox.iso -machine q35
```

After several VMM power up and clean shutdown the kernel faulted. The kernel fault is trigger-able on the specific AMD notebook with vbox5 generic as also vbox5-nova.

alex-ab, 2022-02-09:
@chelmuth, @nfeske : please consider for staging

nfeske, 2022-02-09:
Thank you for the reminder. Merged to staging.

closed 2022-02-15 by nfeske

alex-ab, 2022-04-24:
Unfortunately, on older AMD machines the VM is not working anymore, f00ffe1e6e687f02a8c421e771ceaa3a5d8d0b89 fixes that. I added it to my merge_to_staging branch

commit fba46d51 (2022-02-09): nova: avoid pagefault on invalid SVM exit

Fixes #4391