genodelabs/genode #4392: libc: page fault when using 'Libc::with_libc()' nested from a signal handler

cproc, 2022-01-26:
When I tried to remove the additional entrypoint for handling 'ack_avail' signals from libusb, a page fault occurred during the execution of the signal handler. I couild reproduce this page fault using the `libc_with_libc` test in commit [1].

Log output on base-linux:

```
[init -> test-libc_with_libc]   0x1000000 .. 0x10ffffff: linker area
[init -> test-libc_with_libc]   0x40000000 .. 0x4fffffff: stack area
[init -> test-libc_with_libc]   0x50000000 .. 0x601b1fff: ld.lib.so
[init -> test-libc_with_libc]   0x10e1d000 .. 0x10ffffff: libc.lib.so
[init -> test-libc_with_libc]   0x10d77000 .. 0x10e1cfff: vfs.lib.so
[init -> test-libc_with_libc] --- start with_libc tests ---
[init -> test-libc_with_libc] Start test 1
[init -> test-libc_with_libc] calling with_libc
[init -> test-libc_with_libc] calling with_libc nested
[init -> test-libc_with_libc] Hello from with_libc nested
[init -> test-libc_with_libc] Finished test 1
[init -> test-libc_with_libc] Start test 2
[init -> test-libc_with_libc] calling with_libc
[init -> test-libc_with_libc] calling select with 3 secs timeout from with_libc
[init -> test-libc_with_libc] Finished test 2
[init -> test-libc_with_libc] Start test 3
[init -> test-libc_with_libc] calling with_libc
[init -> test-libc_with_libc] calling with_libc nested
[init -> test-libc_with_libc] calling with_libc nested again
[init -> test-libc_with_libc] Hello from with_libc nested again
[init -> test-libc_with_libc] Finished test 3
[init -> test-libc_with_libc] Start test 4
[init -> test-libc_with_libc] Hello from with_libc nested in signal handler
Kernel: Segmentation fault (signum=11), see Linux kernel log for details
```

dmesg:

```
ld.lib.so[14888]: segfault at 511e8770 ip 0000000010fa5e00 sp 00000000403ff538 error 6 in ds-53 (deleted)[10f9c000+64000]
```

The IP is part of the libc, but not in the code area.

GDB backtrace:

```
#0  0x0000000010fa5e00 in _GLOBAL_OFFSET_TABLE_ () from libc.lib.so
#1  0x0000000010f4a970 in Libc::Kernel::_suspend_main (this=this@entry=0x10fe9350 <Libc::Kernel* unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space>, check=..., timeout_ms=0) at /.../repos/libports/src/lib/libc/internal/kernel.h:380
#2  0x0000000010f53137 in Libc::Kernel::suspend (timeout_ms=<optimized out>, check=..., this=0x10fe9350 <Libc::Kernel* unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space>) at /.../repos/libports/src/lib/libc/internal/timer.h:49
#3  Libc::Main_blockade::block (this=0x10fed658 <Libc::Kernel* unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space+17160>) at /.../repos/libports/src/lib/libc/kernel.cc:34
#4  0x0000000010f4583c in Libc::Monitor::Job::wait_for_completion (this=0x10fed640 <Libc::Kernel* unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space+17136>) at /.../repos/libports/src/lib/libc/internal/monitor.h:134
#5  Libc::Monitor::Pool::monitor (job=..., this=0x10fed528 <Libc::Kernel* unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space+16856>) at /.../repos/libports/src/lib/libc/internal/monitor.h:134
#6  Libc::Kernel::_monitor (this=0x10fe9350 <Libc::Kernel* unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space>, fn=..., timeout_ms=1077933696) at /.../repos/libports/src/lib/libc/internal/kernel.h:553
#7  0x0000000010f6e704 in Libc::Monitor::monitor<Libc::Vfs_plugin::write(Libc::File_descriptor*, void const*, size_t)::<lambda()> > (timeout_ms=0, fn=..., this=<optimized out>) at /.../repos/base/include/util/interface.h:20
#8  Libc::Vfs_plugin::write(Libc::File_descriptor*, void const*, unsigned long) () at /.../repos/libports/src/lib/libc/vfs_plugin.cc:875
#9  0x0000000010ee74cc in __swrite (cookie=<optimized out>, buf=<optimized out>, n=<optimized out>) at /.../contrib/libc-392389f9e1323249c044ba6b7fea6ea3d73100c5/src/lib/libc/lib/libc/stdio/stdio.c:67
#10 0x0000000010ee7722 in _swrite (fp=fp@entry=0x10f9f358 <__sF+312>, buf=buf@entry=0x7ffff7fe3050 "Hello from with_libc nested in signal handler\n", n=n@entry=46) at /.../contrib/libc-392389f9e1323249c044ba6b7fea6ea3d73100c5/src/lib/libc/lib/libc/stdio/stdio.c:119
#11 0x0000000010edd4a5 in __sflush (fp=0x10f9f358 <__sF+312>) at /.../contrib/libc-392389f9e1323249c044ba6b7fea6ea3d73100c5/src/lib/libc/lib/libc/stdio/fflush.c:126
#12 0x0000000010ee1900 in __sfvwrite (fp=0x10f9f358 <__sF+312>, uio=uio@entry=0x403ff9a0) at /.../contrib/libc-392389f9e1323249c044ba6b7fea6ea3d73100c5/src/lib/libc/lib/libc/stdio/fvwrite.c:194
#13 0x0000000010ee5f23 in puts (s=<optimized out>) at /.../contrib/libc-392389f9e1323249c044ba6b7fea6ea3d73100c5/src/lib/libc/lib/libc/stdio/puts.c:69
#14 0x0000000010f4a970 in Libc::Kernel::_suspend_main (this=this@entry=0x10fe9350 <Libc::Kernel* unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space>, check=..., timeout_ms=0) at /.../repos/libports/src/lib/libc/internal/kernel.h:380
#15 0x0000000010f53137 in Libc::Kernel::suspend (timeout_ms=<optimized out>, check=..., this=0x10fe9350 <Libc::Kernel* unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space>) at /.../repos/libports/src/lib/libc/internal/timer.h:49
#16 Libc::Main_blockade::block (this=0x10fed658 <Libc::Kernel* unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space+17160>) at /.../repos/libports/src/lib/libc/kernel.cc:34
#17 0x0000000010f4583c in Libc::Monitor::Job::wait_for_completion (this=0x10fed640 <Libc::Kernel* unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space+17136>) at /.../repos/libports/src/lib/libc/internal/monitor.h:134
#18 Libc::Monitor::Pool::monitor (job=..., this=0x10fed528 <Libc::Kernel* unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space+16856>) at /.../repos/libports/src/lib/libc/internal/monitor.h:134
#19 Libc::Kernel::_monitor (this=0x10fe9350 <Libc::Kernel* unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space>, fn=..., timeout_ms=1077934960) at /.../repos/libports/src/lib/libc/internal/kernel.h:553
#20 0x0000000010f6e704 in Libc::Monitor::monitor<Libc::Vfs_plugin::write(Libc::File_descriptor*, void const*, size_t)::<lambda()> > (timeout_ms=0, fn=..., this=<optimized out>) at /.../repos/base/include/util/interface.h:20
#21 Libc::Vfs_plugin::write(Libc::File_descriptor*, void const*, unsigned long) () at /.../repos/libports/src/lib/libc/vfs_plugin.cc:875
#22 0x0000000010ee74cc in __swrite (cookie=<optimized out>, buf=<optimized out>, n=<optimized out>) at /.../contrib/libc-392389f9e1323249c044ba6b7fea6ea3d73100c5/src/lib/libc/lib/libc/stdio/stdio.c:67
#23 0x0000000010ee7722 in _swrite (fp=fp@entry=0x10f9f358 <__sF+312>, buf=buf@entry=0x7ffff7fe3050 "Hello from with_libc nested in signal handler\n", n=n@entry=21) at /.../contrib/libc-392389f9e1323249c044ba6b7fea6ea3d73100c5/src/lib/libc/lib/libc/stdio/stdio.c:119
#24 0x0000000010edd4a5 in __sflush (fp=0x10f9f358 <__sF+312>) at /.../contrib/libc-392389f9e1323249c044ba6b7fea6ea3d73100c5/src/lib/libc/lib/libc/stdio/fflush.c:126
#25 0x0000000010ee1900 in __sfvwrite (fp=0x10f9f358 <__sF+312>, uio=uio@entry=0x403ffe90) at /.../contrib/libc-392389f9e1323249c044ba6b7fea6ea3d73100c5/src/lib/libc/lib/libc/stdio/fvwrite.c:194
#26 0x0000000010ee5f23 in puts (s=<optimized out>) at /.../contrib/libc-392389f9e1323249c044ba6b7fea6ea3d73100c5/src/lib/libc/lib/libc/stdio/puts.c:69
#27 0x0000000010f4aa22 in Libc::Kernel::_user_entry (kernel=0x10fe9350 <Libc::Kernel* unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space>) at /.../repos/libports/src/lib/libc/internal/kernel.h:308
#28 0x0000000010f4af72 in call_func (arg=0x10f4a61d <Libc::Kernel::_suspend_main(Libc::Suspend_functor&, unsigned long long)+157>, func=0x10f4a9e0 <Libc::Kernel::_user_entry(Libc::Kernel*)>, sp=0xc9840ab60f0774f0) at /.../repos/libports/src/lib/libc/spec/x86_64/internal/call_func.h:26
#29 Libc::Kernel::run (this=0x10fe9350 <Libc::Kernel* unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space>, app_code=...) at /.../repos/libports/src/lib/libc/internal/kernel.h:425
#30 0x0000000000000000 in ?? ()
```

[1] https://github.com/cproc/genode/commit/862de0f4a5d38cbacff2868df1e0e9b7d5e0253a

chelmuth, 2022-01-26:
I'm wondering it it really makes sense to use the `Io_signal_handler` in the Libc component here. In some cases this means the main thread is suspended in a blocking operation (e.g., read) but will execute the signal handler in nested fashion, which will then call libc function itself. If the blocked operation and the nested function need the same resource (e.g., a mutex in stdio) we get a deadlock, in other cases even worse results. More and more I get the feeling the libc component is reshaping from a neat tool into a can of worms...

nfeske, 2022-01-27:
@chelmuth You are spot-on about the `Io_signal_handler`. As a rule, an I/O signal handler must not have global side effects. The call of `with_libc` is a potentially far-reaching global side effect. Hence, such calls cannot be placed in an I/O signal handler. @cproc, coming back the libusb, do you see a chance to decouple the interaction with the libc from the I/O signal handler?

Btw, just yesterday, while cleaning up the aftermath of #4390, I came across another instance where this rule is violated. The rump_vfs plugin calls this ( https://github.com/genodelabs/genode/blob/master/repos/dde_rump/src/lib/vfs/rump/vfs_rump.cc#L44) function from an I/O signal handler (triggered by a timer connection). The function calls right into the rump-kernel code. This just happened to work (by chance?) because the blocking operations of the packet stream implicitly used to supress _all_ signals including I/O signals.

cproc, 2022-01-28:
libusb calls `poll()` to wait for completion events, so the signal handler at least needs to write into a pipe to unblock the `poll()` call. Commit 639388c reduces the complexity of the signal handler to this wake-up code, which at least has the positive effect that the pthread registration is not needed anymore. Debug commit 3714161 removes the separate entrypoint, but causes the page fault (for reference).

cproc, 2022-02-01:
Commit b642b83 implements the VFS plugin idea. Seems to work well so far.

nfeske, 2022-02-02:
> Seems to work well so far.

That's very good to hear. Thank you for implementing the idea. Now I wonder, is the mutex you introduced in the first fix still needed?

nfeske, 2022-02-02:
Super!

nfeske, 2022-02-15:
Fixed in master.

closed 2022-02-15 by nfeske