genodelabs/genode #4393: base: `Heap::free()` releases entire dataspace if pointer does not point to the beginning of the allocated block

jschlatow, 2022-01-27:
When working on a vfs plugin, I implemented a class with two base classes one of which being Single_vfs_handle. On component destruction, I encountered a page fault and looked into it. As it turned out, I messed up the inheritance order: I initially had used Single_vfs_handle as a second base class. Switching the order of the two base classes fixed this issue, but I was still wondering about why this wasn't handled more gracefully.

Turned out, `Heap::free()` determines whether a to-be-freed address is managed by its local allocator by calling `size_at()`. If it returns 0, it searches a dataspace to which the address belongs and frees the latter entirely. Any subsequent access to any other object that resided in this dataspace will thus cause a page fault.
I noticed that this behaviour might be a regression introduced by commit b78e47b. If I revert the commit, my scenario fails more gracefully with the following error:

```
Error: virtual void Genode::Allocator_avl_base::free(void*): given address (0xadc8) is not the block start address (0xaa70)
```

Alright, I am wondering if reverting the commit would break #4325, which introduced this change.
@cnuke what was the motivation behind it?

Despite this there is still a remaining question though. As explained by the comment in `Genode::destroy()`, the delete operator should be able to magically determine the actual base pointer of the class via the vtable. Yet, it apparently fails to achieve this in my scenario. Any ideas on that?

jschlatow, 2022-01-27:
I added commit 8e9324e, which triggers the observed behaviour in the new_delete test.

jschlatow, 2022-01-28:
I like the idea of applying the `Attempt` pattern to `free()`. I think this may also render the `need_size_for_free()` interface unnecessary. @nfeske, shall I make an attempt (pun intended)?

>Despite this there is still a remaining question though. As explained by the comment in Genode::destroy(), the delete operator should be able to magically determine the actual base pointer of the class via the vtable. Yet, it apparently fails to achieve this in my scenario. Any ideas on that?

I think I have an explanation for that. Deallocation of virtual base classes is apparently not completely handled by the standard. Yet, some smart people came up with [a solution](https://stackoverflow.com/questions/44558119/why-do-i-have-two-destructor-implementations-in-my-assembly-output) that is part of the "Itanium ABI". In brief summary, here is what happens: Since the delete operator is global, there is no way to make it virtual on its own. The workaround is, however, that the compiler generates two destructors: a (standard) object destructor and a deleting destructor. Whenever the delete operator is called on a virtual class, the compiler calls the deleting destructor instead of the standard destructor.

As it appears, this solution is not effective in the Genode code base since `Genode::destroy()` calls an overloaded version of the delete operator. Nevertheless, I found out that there is a version of `dynamic_cast` for polymorphic objects by which we can determine the most address of the most derived object. The result is found in commit 92ddfc1, which is accompanied by commit 0867030 for test coverage.

Out of curiosity, I also had a look at the assembly code and noticed that the compiler indead generates two destructors for every virtual class. Given the fact that the deleting destructor is never called, I looked for compiler flags to disable this. I found `-fdeclone-ctor-dtor` which is part of `-Os` and which should generate a common implementation instead of cloning the functions. Yet, the flag had no effect on my binary.

nfeske, 2022-01-31:
> I like the idea of applying the Attempt pattern to `free()`

I was actually not thinking of `free` but the replacing the return value of `Allocator_avl::size_at` (https://github.com/genodelabs/genode/blob/master/repos/base/include/base/allocator_avl.h#L290). Right now, the return value is just `size_t` whereas 0 is a special value. But as there are two different conditions that can result in 0 (see https://github.com/genodelabs/genode/blob/master/repos/base/include/base/allocator_avl.h#L290 lines 403 and 405).

To distinguish both conditions, I suggest changing the return value from `size_t` to somthing like:

```
enum class Size_at_error {
  UNKNOWN_ADDR,      /* no allocation at specified address */
  MISMATCHING_ADDR,  /* specified address is not the start of a block */
};

using Size_at_return = Attempt<size_t, Size_at_error>;
```

With this change, the checks at the heap (https://github.com/genodelabs/genode/blob/master/repos/base/src/lib/base/heap.cc#L261) and lx_kit (https://github.com/genodelabs/genode/blob/master/repos/dde_linux/src/lib/lx_kit/memory.cc#L176) can test for either one of the specific conditions.

> shall I make an attempt (pun intended)?

I would appreciate that. Note that you may need to adjust the ABI (base/lib/symbols/ld) regarding `size_at`.

Regarding the delete magic, I unable to add much substance, but like to share my sentiment that we should better try to avoid dynamic casts in the base framework. Down the road - should we succeed with eliminating C++ exceptions from the base framework - we could drop the dependency from the C++ runtime (libsupc++, libeh) thereby reducing the TCB complexity further. However, this also excludes runtime type information (rtti), which is a prerequisite for dynamic casts. Following this long-term ambition, I think we should try to avoid relying on rtti whenever possible.

jschlatow, 2022-01-31:
> I was actually not thinking of `free` but the replacing the return value of `Allocator_avl::size_at`

That's of course way easier :wink: Thanks for the perfectly clear explanation.

> Regarding the delete magic, I unable to add much substance, but like to share my sentiment that we should better try to avoid dynamic casts in the base framework. Down the road - should we succeed [...]

That's a very good point. Given that fixing `size_at` will bring back an appropriate error message, I can live without the delete magic. In my scenario, I can simply make Single_vfs_handle the primary base class to remove the error message.

In other cases, the dynamic cast could be performed when calling `Genode::destroy()`. An alternative idea is to allow for the definition of an optional virtual delete operator that is called by `Genode::destroy()` if present. However, this is probably a non-issue since I guess these are very rare cases anyway.

jschlatow, 2022-01-31:
Commit bf63eb5 replaces the return value of `Allocator_avl::size_at`. @nfeske, is this what you had in mind?

nfeske, 2022-02-01:
> is this what you had in mind?

That's already quite good. But you missed a few "features" of the `Attempt` utility that can make the core more succinct:

The `Attempt` utility is one of the rare cases that benefit from C++'s implicit constructor calls, which allow you to avoid the repetitive use of `Size_at_result` when returning values. E.g.,

I would change the following code:
```
	if (b && (b->addr() != (addr_t)addr))
		return Size_at_return(Size_at_error::MISMATCHING_ADDR);

	return (b && b->used()) ? Size_at_return(b->size())
	                        : Size_at_return(Size_at_error::UNKNOWN_ADDR);
```
to the following:
```
	if (b && (b->addr() != (addr_t)addr))
		return Size_at_error::MISMATCHING_ADDR;

	if (b && b->used())
		return b->size();

	return Size_at_error::UNKNOWN_ADDR;
```
It may be subjective, but the latter version would be less noisy to my eyes.
You may review your patch, search for `Size_at_result` and try removing it where possible. Please don't be shy with replacing the use of the ternary `?` operator by two distinct `return` statements where it makes the code easier to read.

The following code got quite a bit inflated by your change (at https://github.com/genodelabs/genode/commit/bf63eb5bb7e52eec1c870744d8a2028c7b827c88#diff-38afddfa5160cce691e0af97c90034d23f1a899d17228f97cbdcd25b744ce6c4L259):

```
       size_t const size = _alloc->size_at(addr);

       if (size != 0) {

               /* forward request to our local allocator */
               _alloc->free(addr, size);
               _quota_used -= size;
               return;
       }
```

Here, the `Attempt::ok()` and `Attempt::failed()` can come to the rescue, allowing you to largely preserve the original structure:
```
	Allocator_avl::Size_at_return const size_at_return = _alloc->size_at(addr);

	if (size_at_return.ok()) {

		/* forward request to our local allocator */
		size_at_return.with_result(
			[&] (size_t size) {
				_alloc->free(addr, size);
				_quota_used -= size; },
			[&] (Allocator_avl::Size_at_error) { });

		return;
	}
```
You may review the other parts of the patch that may benefit from the use of these accessors, in particular in cases of early return. In some cases, the key to simplicity is the combination of `Attempt::failed()` and `Attempt::with_error()`, like so:

```
	using Size_at_error = Allocator_avl_base::Size_at_error;

	if (size_at_result.failed()) {
		size_at_result.with_error([&] (Size_at_error e) {

			if (e == Size_at_error::MISMATCHING_ADDR)
				error(...);
		});
		return;
	}
```

...yes, the game is a bit like Sudoku.

As a matter of consistency, I'd like to stick to naming the attempt type as `Size_at_result` instead of `Size_at_error`, following the existing naming of `Alloc_result`. I'm sorry that my original sketch was bringing you on the wrong track here. Please forgive.

Instead of using `auto` you may consider specifying the actual type. If this becomes too noisy, it often helps to define an alias (for example `using Error = Allocator_avl_base::Size_at_error`). This way, the code stays self-documenting, e.g., when seeing a line like:
```
  [&] (Error) { }
```
it becomes immediately clear that a potential error is ignored. This is not so obvious when using `auto`.

I hope you don't mind my nitpicking. ;-)

jschlatow, 2022-02-01:
> I hope you don't mind my nitpicking. ;-)

Not at all. Thanks for the detailed feedback.

I used `convert` because I need to decide, depending on the error type, whether to return from `Heap::free()` and find this more readable than capturing a local variable and modifying it in the lambda. With your proposal, I would end up with something like:

```
using Size_at_error = Allocator_avl_base::Size_at_error;

Allocator_avl::Size_at_result const size_at_result = _alloc->size_at(addr);

if (size_at_result.ok()) {
      /* forward request to our local allocator */
      size_at_result.with_result(
	      [&] (size_t size) {
		      _alloc->free(addr, size);
		      _quota_used -= size; },
	      [&] (Allocator_avl::Size_at_error) { });
      
      return;
}
else {
      bool abort = false;
      size_at_result.with_error([&] (Size_at_error e) {
      
	      if (e == Size_at_error::MISMATCHING_ADDR) {
		      error(...);
                      abort = true;
              }
      });
      if (abort) return;
}
```

How do you feel about adding a `has_error` accessor or an `operator ==` to the Attempt class so that one can check for a certain error type in a single line of code? This would remove a lot of noise from the _else_ case.

nfeske, 2022-02-01:
You can drop the `else` because the `if` body always returns. But that's beside the actual point you are raising. ;-)

To your actual question: yes, I think that's a sensible suggestion. I like it.

Regarding the flavor (`has_error` vs. `operator ==`), even though I am generally not a friend of defining custom operators, in this case, the `operator ==` would be a tasteful addition. I can imagine that it would make the code quite nice to read. Would you want to give it a go? If so, please let us put this change of `util/attempt.h` into a separate commit.

jschlatow, 2022-02-01:
Commit 68021d0 contains the change to _util/attempt.h_. I also noticed that the lambda for the error case was passed by value. As I suppose this was not intentional, I changed this with commit 282289b.

The resulting changes of `Allocator_avl` are found in commit b2e87a8.

nfeske, 2022-02-01:
This looks excellent! Thanks for the improvements, which I merged to staging just now.

nfeske, 2022-02-15:
Fixed in master.