genodelabs/genode #4394: vfs: implement plugin for tap device emulation

jschlatow, 2022-01-27:


jschlatow, 2022-01-27:
Commit 3f78081 implements a basic vfs_tap plugin. By default the plugin uses a Nic_session. This can be optionally changed to an Uplink_session to provide a user assignable MAC address. Details can be found in the README.

@chelmuth, if you want to test the plugin with virtualbox, you probably need to provide a few dummy functions. FreeBSD's API for tap devices is quite a bit different than the Linux API and, AFAIK, virtualbox uses the latter. As described in the README file, a few dummy defines are sufficient for simple scenarios though.

chelmuth, 2022-01-27:
Can't wait to test this next week in Dresden :-)

nfeske, 2022-01-28:
This looks absolutely splendid!

Your code stays true to the spirit of the terminal plugin, e.g., following the compound-directory pattern and providing the meta information as individual files as well as an XML info file. Also your README - pointing out the practical differences between the FreeBSD and Linux interfaces and the hints about how to deal with that - is excellent.

While browsing though the commit, I wondered if the there speaks anything against hosting the VFS plugin in the os repository instead of libports? As far as I can see, the plugin only depends on the base, vfs, and os APIs. The test should of course to remain in libports as it depends on the libc. @jschlatow What do you think?

jschlatow, 2022-01-28:
@nfeske thanks for the feedback.

There is no technical reason against putting the plugin into os. I still hestitated, because a) I expect it to be used only in conjunction with libc and b) the plugins currently in os are all built-in in contrast to the shared library plugins found in other repos. I'd be fine with both though.

nfeske, 2022-01-28:
>  I expect it to be used only in conjunction with libc

That's true, but this reasoning would also apply for the terminal-VFS-plugin, doesn't it?

I tend to draw the line by looking at dependencies. Code specifically developed for Genode and which does not depend on any 3rd-party library qualifies for the os repository.

> the plugins currently in os are all built-in in contrast to the shared library plugins found in other repos

Unless we decide to add the tap plugin to the built-in feature set of the vfs.lib.so, we can opt to host its source under os/src/lib/vfs/tap/ and build it as a shared library. I think that's not a contradiction.

That said, if you feel better keeping it at libports, that's completely fine by me, too.

jschlatow, 2022-01-28:
Agreed :smile:. Commit 5e0a2c3 moves the plugin to the os repository.

jschlatow, 2022-02-03:
@nfeske, when drafting my genodians article about the vfs_tap I noticed two things about the `File_io_service` interface:

1. I cannot find a place where the `check_unblock()` method is actually used. I assume this might be a relict.
2. The `Single_file_system` is a convenient helper for quickly implementing a special-purpose file. It forwards most of the `File_io_service` methods to the handle, including the `read_ready()` method. I think it would be nice to also forward the `notify_read_ready()` method to the handle by default since this is closely related to `read_ready()`. Yet, as far as I can see, only the terminal plugin and the tap plugin would benefit from that.

nfeske, 2022-02-04:
> I cannot find a place where the check_unblock() method is actually used. I assume this might be a relict.

I think you are right. It is from a time when the VFS resided in the noux runtime - long ago. If you like, please try to remove it. I would welcome a patch for sure. :-)

> I think it would be nice to also forward the notify_read_ready() method to the handle by default since this is closely related to read_ready(). Yet, as far as I can see, only the terminal plugin and the tap plugin would benefit from that.

No objection. Please go ahead. If we can simplify the plugins, I'm all for it.

nfeske, 2022-02-28:
The vfs_tap plugin is featured in Genode 22.02.

closed 2022-02-28 by nfeske