genodelabs/genode #4395: readlink() returns wrong value on success

ttcoder, 2022-01-28:
Looks like a small deviation from standard, but presumably significant enough to warrant a fix in libc VFS plug-in.

Reproducible with e.g.
```
#include <unistd.h>
..
symlink( "/", "/boot/symlink-to-slash" );
char buf_readlink[64] = {};
if( strlen("/") != readlink( "/boot/symlink-to-slash", buf_readlink, sizeof(buf_readlink) ) )
  Genode::error( "readlink() ought to return 1, but returns 64 instead" );
```

References:
- did some quick testing in haiku with a two-line .c file, readlink() did return 1 there indeed.
- https://www.unix.com/man-page/freebsd/2/readlinkat/ : 
> The call returns the count of characters placed in the buffer if it succeeds

We could add a test like this (in libports/src/test/libc_vfs/main.cc), to check for future regressions:

```
if ((symlink("/", "symlinks_supported") == 0) || (errno != EPERM)) {
  CALL_AND_CHECK(ret, symlink("../a", "c/d"),
+  char buf_readlink[64] = {};
+  CALL_AND_CHECK(ret, readlink("c/d", buf_readlink, sizeof(buf_readlink), ret==strlen("../a")),
```

(ran into this during my libbe.so port, BSymlink::MakeLinkedPath() does not like the readlink() return code returned by Genode's libc and itself raises an error, even though the symlink value did get read into the buffer correctly -- still blazing my way through my port and having a good time :-).

nfeske, 2022-02-08:
Thanks for posting, @ttcoder.

I tried reproducing this issue with commit 752a299b587433b67acbb444335a98db0292b301 but the test succeeds when executing it via

```
x86_64$ make run/test PKG=test-libc_vfs KERNEL=linux BOARD=linux
```

I hope that I understood your description correctly but I'm not 100% sure. Can you bring me on the right track with reproducing the problem?

ttcoder, 2022-02-08:
My bad -- the wrong length actually came all the way down from my VFS plug-in for NTFS, which I had configured the test for. Quickly realized that, after reverting the test to use a 'ram' VFS node, and seeing that readlink() was now correctly returning '1' instead of '64' indeed.

Fixed the problem in my VFS plug-in and now it passes all tests.

Closing this invalid ticket.
I leave the newly added test to your consideration, i.e. is it worth it to leave it in place or not etc.

Sorry for the noise and thanks!

closed 2022-02-08 by ttcoder