genodelabs/genode #4397: Port of the Linux Wireguard implementation

m-stein, 2022-02-01:
The initial approach would be to port Wireguard as a dedicated Genode component that requests a NIC session at the public side (the UDP-tunnel side) and an Uplink session at the the private side (the plain-packet side). Each of the two sessions could be provided by a NIC router in order to connect the component with a "real" uplink respectively the VPN-internal applications. The configuration of the Wireguard device, that is normally done through a user application and Linux Netlink, can be done via the XML component configuration.

m-stein, 2022-02-01:
There is a first prototype of the port on my ``genodelabs/wireguard`` topic branch. It is accompanied by 3 run scripts ``wg_ping_inwards``, ``wg_ping_outwards``, and ``wg_lighttpd`` that demonstrate that the prototype is already capable of handling simple network traffic. Now, I'd do a brief clean-up- and documentation step before I go on exposing the port to more challenging scenarios.

m-stein, 2022-05-18:
@chelmuth This is ready for merging:
228d16acd0 dde_linux: port of WireGuard kernel module
82726113c0 virt_linux: support arm_64
2b9042b614 lx_emul & arm: define ZERO_PAGE
c1b001e52a lx_emul: fix unaligned size arg on DMA alloc
fcfd21577c net: add methods required for WireGuard port
9ca146dc4a net: fix conversion compiler errors
f267afc167 nic_router: fix bad gateway values in README

m-stein, 2022-05-19:
@chelmuth Thanks for your thorough review! I'll apply your remarks today.

m-stein, 2022-05-19:
@chelmuth Here are some first patches:
66948e2c19 fixup "dde_linux: port of WireGuard kernel module" (commented code)
dac3c85986 lx_emul & x86: no unnecessary memset of zero page
70c989baf0 fixup "lx_emul & arm: define ZERO_PAGE" (no memset)
85ced3e0c9 fixup "net: add methods required for WireGuard port" (const args)

I'll fix the local_submit and entropy stuff later today.

chelmuth, 2022-05-19:
With the current patches on staging I still got the following.

```
.../genode/repos/dde_linux/src/lib/lx_emul/shared_dma_buffer.cc: In function ‘genode_shared_dataspace* lx_emul_shared_dma_buffer_allocate(long unsigned int)’:
.../genode/repos/dde_linux/src/lib/lx_emul/shared_dma_buffer.cc:25:9: error: ‘align_addr’ was not declared in this scope; did you mean ‘Genode::align_addr’?
```

Applying the compiler hint helped.

```diff
--- a/repos/dde_linux/src/lib/lx_emul/shared_dma_buffer.cc
+++ b/repos/dde_linux/src/lib/lx_emul/shared_dma_buffer.cc
@@ -22,7 +22,7 @@ struct genode_shared_dataspace : Lx_kit::Dma_buffer {};
 extern "C" struct genode_shared_dataspace *
 lx_emul_shared_dma_buffer_allocate(unsigned long size)
 {
-       size = align_addr(size, 12);
+       size = Genode::align_addr(size, 12);
        Lx_kit::Mem_allocator::Buffer & b = Lx_kit::env().memory.alloc_buffer(size);
 
        /*
```

Should we change 66ca1f7100154855133da93545cbb2428cfd7b45 accordingly?

skalk, 2022-05-19:
@chelmuth @m-stein the align_addr here is actually superfluent. The memory allocator of the Lx_kit already fulfills this constraint when alloc_buffer is called. I would simply remove that commit.

chelmuth, 2022-05-19:
> @chelmuth @m-stein the align_addr here is actually superfluent. The memory allocator of the Lx_kit already fulfills this constraint when alloc_buffer is called. I would simply remove that commit.

Done in a13914b0d609ffa1b9a3a0c97af25767815b6e6c

m-stein, 2022-05-19:
@chelmuth Here are the missing patches for your suggestions:
84837eb10c lx_emul/random: reseed the sponge regularly
8b0e1ec6cc lx_emul & wireguard & wifi: centralized random.cc
0cfd529c81 fixup "dde_linux: port of WireGuard kernel module" (ip config notifier)

The reseeding commit is only a suggestion of mine to meet your point about reseeding. Is this what you had in mind?

m-stein, 2022-05-19:
In a quick test via wg_fetchurl on a 10mb transfer load I could observe that the randomness commits seem to raise the throughput of WireGuard like 300%.

m-stein, 2022-05-20:
@chelmuth I've added three fixups for the lx_emul randomness:
3c70f68eff squash "lx_emul/random: reseed the sponge regularly" (significant clean-up)
11ed5b98cf lx_emul/random: reseed the sponge regularly
94ff1e0c37 fixup "lx_emul & wireguard & wifi: centralized random.cc" (rename Xoroshiro)
0eabb804af squash "lx_emul & wireguard & wifi: centralized random.cc" (security warning)
d3ea0a7b32 lx_emul & wireguard & wifi: centralized random.cc

m-stein, 2022-05-20:
The semantics of the code are not changed by the fixups.

nfeske, 2022-05-25:
@m-stein now that the first version entered the master branch, shall we close this issue? I think that any subsequent developments would deserved separate issues.

m-stein, 2022-05-26:
@nfeske I think the same.

closed 2022-05-26 by m-stein