genodelabs/genode #4405: Remove `http_block`, `block_cache`, and `test/block`

nfeske, 2022-02-04:
Those components rely on the old `block/component.h` framework classes, which have long been superseded by the `Block::Request_stream`. In order to dispose the legacy block/component.h API, those components would need to be updated. However, given that they are lingering unused since years, I doubt that the effort is justified.

Should the functionality become needed, we would probably want to implement those features differently, anyway. E.g., supporting SSL for accessing a block device over HTTP, or implementing a block cache as VFS plugin instead of a bump-in-the-wire server. So I don't see a revival of the existing components very likely.

I presume that test/block has been superseded by app/block_tester and vfs_block in the meantime.

@skalk and @cnuke, what is your opinion?

cnuke, 2022-02-04:
@nfeske I am fine with removing them.

nfeske, 2022-02-15:
Fixed in master.

closed 2022-02-15 by nfeske