genodelabs/genode #4406: jpeg.port: update sha hash

m-stein, 2022-02-07:
I don't know why (and indeed it seems suspicious to me) but the SHA of the JPEG download doesn't work anymore for me (genodelabs/staging). I've created a simple fix but am unsure whether we should investigate the cause of the regression.

m-stein, 2022-02-07:
07f12671a1 jpeg.port: update sha hash

chelmuth, 2022-02-07:
I checked the contents by

```
tar --one-top-level --strip-components=1 -xf jpegsrc-old.v9d.tar.gz
tar --one-top-level --strip-components=1 -xf jpegsrc-new.v9d.tar.gz
diff -ur jpegsrc-old.v9d jpegsrc-new.v9d

tar -tvf  jpegsrc.v9d-old.tar.gz | sort -k6 > old.lst
tar -tvf  jpegsrc.v9d-new.tar.gz | sort -k6 > new.lst
diff -u old.lst new.lst
```

and merged the slightly adjusted f928570a625ca1c18782334ea406c3daad1ca0b6.

commit f928570a (2022-02-07): jpeg: update hash of downloaded archive

The archive contents are equivalent except the creation date of the
top-level directory.

  -drwxr-xr-x guivol/users      0 2021-01-07 12:54 jpeg-9d/
  +drwxr-xr-x guivol/users      0 2022-01-04 12:02 jpeg-9d/

Fixes #4406
closed 2022-02-15 by nfeske