genodelabs/genode #4408: mk: omit transitive shared-library dependencies from linker command line

nfeske, 2022-02-08:
Right now, binaries created directly within a build directory are linked slightly different compared to binaries created as depot archive. When created in the build directory, all shared-library dependencies including transitive shared-library dependencies of the target's used shared libraries end up at the linker command line. In contrast, when building a depot archive - where transitive shared libraries are not known because they are hidden behind the library's ABI - only the immediate dependencies appear at the linker command line. To improve the consistency, we should better link without transitive shared objects in both cases.

nfeske, 2022-02-08:
Fixed by https://github.com/genodelabs/genode/commit/9a320a5fbbb8d6e829c94d8c774114e533abeea8

ssumpf, 2022-02-10:
@nfeske: I will change Java for this.

chelmuth, 2022-02-11:
> @nfeske: I adapt Java to this change.

genodelabs/genode-world@b9c32477110612e7fe532831539959e86a544dbe

ssumpf, 2022-02-11:
> > @nfeske: I adapt Java to this change.
> 
> [genodelabs/genode-world@b9c3247](https://github.com/genodelabs/genode-world/commit/b9c32477110612e7fe532831539959e86a544dbe)

:+1:

nfeske, 2022-02-15:
Fixed in master.

closed 2022-02-15 by nfeske